blackmagic-debug/blackmagic #76: LPC21xx support

maknoll, 2015-01-28:
I would like to write support for the LPC21xx ARM7TDMI family. Maybe you could give me some hints what has to be implemented?

gsmcmullin, 2015-01-29:
Hi Martin

To start, the support for the ARM7TDMI support will need to be finished.
There is an unfinished driver here:
https://github.com/blacksphere/blackmagic/blob/master/src/arm7tdmi.c

I'm no longer working with any ARM7 devices, so have had little motivation
to finish this.
At the very least the functions to read and write memory need to be
implemented.  The function pointers need to be filled in the target
structure in arm7tdmi_jtag_handler().
You'll need to check the ARM literature for how to implement these
functions.

Once the core functions are all implemented, an LPC21xx driver will need to
be written that provides the XML memory map to GDB and implements flash
programming routines.

Cheers,
Gareth

maknoll, 2015-06-29:
Thanks for your support. Since i don't know if and when i will finish this endeavor and it is not a real issue i'm closing it.

closed 2015-06-29 by maknoll