blackmagic-debug/blackmagic #88: stuck at DFU mode

hackrid, 2015-04-27:
Hi,

I'm trying to flash the debugger portion of an STM32 Nucleo board with the BMP binaries.
But it's stuck in the upgrade mode even though I flashed the blackmagic.bin at 0x08002000.

I once got past the upgrade mode, but can't remember what I did to acomplish that.

what prevents the bootloader to start the BMP application?

gsmcmullin, 2015-04-27:
There is no official support for running on the Nucleo board, so I have no idea what binaries you're using or what they may be doing.  The official bootloader will jump to the application if the application's initial stack pointer is somewhere in ram and the DFU request button isn't held down at boot time.

closed 2015-04-27 by gsmcmullin

hackrid, 2015-04-28:
the clear this a bit: i think i forgot to revert the solder bridges back to its original state. this causes the bootloader never to jump to the application.