blackmagic-debug/blackmagic #89: Mass Erase on F4 fails

hackrid, 2015-04-29:
I'm trying to erase a cortex m4 controller but it constantly fails

```
[peter@localhost ~]$ arm-none-eabi-gdb
GNU gdb (GDB) 7.8.2
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon version
Black Magic Probe (Firmware v1.6-rc0-50-g482070c-dirty)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
(gdb) mon swdp_scan
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      STM32F4xx
(gdb) att 1
Attaching to Remote target
0x08000f22 in ?? ()
(gdb) mon erase_mass
Erasing flash... This may take a few seconds. -Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

gsmcmullin, 2015-04-30:
It's working for me.  Tested with master on native hardware, STM32F405RG target, Linux host.

UweBonnes, 2015-05-01:
Probably the device is protected. The option bits should be checked and eventually reset.

hackrid, 2015-05-04:
thanks Uwe for you tip.
tryed to erase the option bytes. no success.
any more ideas?

```
peter@localhost ~]$ arm-none-eabi-gdb
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon swdp_scan
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      STM32F4xx
(gdb) att 1
Attaching to Remote target
0x08000f5e in ?? ()
(gdb) mon option
usage: monitor option erase
usage: monitor option write <value>
0x1FFFC000: 0xAAEF
0x1FFFC008: 0x3FFF
(gdb) mon option erase
0x1FFFC000: 0xAAEF
0x1FFFC008: 0x3FFF
(gdb) mon erase_mass
Erasing flash... This may take a few seconds. /Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

UweBonnes, 2015-05-04:
You need to hard reset (power reset) the device so that the no option fuse settings take effect.

hackrid, 2015-05-11:
I have erased the option bytes using the stm32unlock tool from Segger with one of their debuggers. the option bytes did not change! the device was never protected in any way.

mass erasing the device still fails...

UweBonnes, 2015-05-11:
I just tried with blacksphere GIT head and a F429 discovery and it worked as expected. Before erase a program was running.
(gdb) mon erase_mass
Erasing flash... This may take a few seconds. -
(gdb) x 0x08000000
0x8000000:      0xffffffff
Aferwards no program is running.

How does mass erase fail? Why do you think that it fails?

hackrid, 2015-05-12:
i'm getting these strange error messages:
```
Erasing flash... This may take a few seconds. -Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing..
```

UweBonnes, 2015-05-12:
This looks like the (JTAG/SWD?) communication failing.  Is this a custom board? Does the STM32F4 eventually provide a buffer enable in the JTAG chain? And with erase this buffer enable is missing, tearing the chain apart?

hackrid, 2015-05-13:
i would also guess swd / jtag comm. failure.
I'm trying to erase the F4 on a STM32F429 Discovery board using the STM32F4 from a STM32F4 Discovery. Just a few jumper wires between them

UweBonnes, 2015-05-16:
Try with more ground wires. Signal integrity and chance of success improved with the number of wires.

gsmcmullin, 2015-05-17:
Sorry for the delay.  I don't believe this is a simple communication
fault.  Such should be detected and reported to GDB as an error.

We would need to connect a debugger to the BMP in question and find out
where it's spinning. Do you have another BMP or other means of interrupting
the stuck BMP and getting out a backtrace?

I can't debug this as I can't reproduce the problem.  It's working
perfectly for me here.

chaosAD, 2016-11-19:
Your target voltage is absent:
....
Target voltage: ABSENT!
....

esden, 2017-02-23:
Is there a conclusion on this, or can we close this bug report?

esden, 2017-04-12:
As we can not reproduce this issue I am moving it to the V1.7 milestone.

Arrowana, 2017-09-28:
I am using a STM32F439 and experiencing the same problem
When I run mon erase_mass the BMP hangs and requires a hard reset to work again

```
Available Targets:
No. Att Driver
 1      STM32F42x
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x1fff084c in ?? ()
(gdb) mon erase_mass
Erasing flash... This may take a few seconds. /Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```
The command does erase the flash, however running it will always result in this error.
So I believe this isn't an option byte issue.

UweBonnes, 2017-09-28:
Pierre,

you could be hit by the gdb timeout bug, https://github.com/blacksphere/blackmagic/issues/181

esden, 2021-12-15:
I think the most likely explanation is that this was caused by the GDB timeout bug. As this issue is very old by now, so I am closing it. If this is still present feel free to reopen this issue.

closed 2021-12-15 by esden