blackmagic-debug/blackmagic #90: Current master fails to flash stm32f0discovery (stm32f051)

maknoll, 2015-06-12:
it works on v1.6-rc0

i will test this on a native probe, but i assume it will be the same result.

    No. Att Driver
     1      STM32F0
    0xfffffffe in ?? ()
    (gdb) load
    Loading section .isr_vector, size 0xc4 lma 0x8000000
    Error writing data to flash
    (gdb) detach

UweBonnes, 2015-06-22:
The flashstubs are compiled for M3. STM32F0 is a M0. Compiling the flashstubs for M0 makes it work again.  I have some patches, including this one pending on https://github.com/UweBonnes/blackmagic.git

maknoll, 2015-06-22:
That makes sense, thanks! Do you wanna make a pull request?

UweBonnes, 2015-06-22:
Martin> That makes sense, thanks! Do you wanna make a pull request?  —
    Martin>

Success report before submitting the pull request would help :-)
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 162516 -------- Fax. 06151 164321 ----------

maknoll, 2015-06-22:
Oh, sure :) I don't have the board here now, but i will let you know once i have tested it.

maknoll, 2015-06-22:
I have cherry-picked your 3 latest commits an they solve the issue (actually the latest alone does, but i liked the other ones too).

Tested on a STM32F072, not tested on a M3, but since ARMv6 is a subset of v7 it should be allright.

    (gdb) load
    Loading section .text, size 0x1ec lma 0x8000000
    Start address 0x8000170, load size 492
    Transfer rate: 9 KB/sec, 492 bytes/write.
    (gdb) compare-sections 
    Section .text, range 0x8000000 -- 0x80001ec: matched.

gsmcmullin, 2015-06-30:
@UweBonnes Thanks for the observation.  This is fixed by #92

commit 762e5406 (2015-06-30): Merge pull request #92 from maknoll/master

Fixes #90 flashing of STM32 m0 chips
closed 2015-06-30 by gsmcmullin