blackmagic-debug/blackmagic #103: Target is halted on powerup

rogerclarkmelbourne, 2015-11-09:
Hi,
I guess this isn't an issue for everyone, but I'm using the BMP with a Nordic nRF51822 and when I power up, the BMP halts the nRF51

So I either have to run GDB, connect to BMP, scan for the target, attach to the target, the run the target.

Ideally, I'd like to be able to compile a version of the BMP that doesn't halt the target when its power on.

Can anyone advise how I can go about this.

Note. I've already recompiled the code as the board I'm using for the BMP has a different clock crystal to any of the other platforms, actually its a GD32 which is very similar to the STM32 and in my case it happens to have a 12Mhz instead of an 8Mhz Crystal

So I'm quite familiar with building and modifying code, but the BMP code is quite extensive and I'm not sure where to start looking

Thanks

rogerclarkmelbourne, 2015-11-09:
Update.
I'm not sure if this is the correct way to do it, but I've found if I change the current j66_coreboard code so that it doesnt clear the SWD or SDIO pins on initialisation

This seems to get over my problem

I'm not sure if clearing these pins is normal practice on the BMP, I'll need to review the other platform's init code.

Still it would be good to know if this is a good or bad thing to do.

esden, 2017-02-23:
Is this a resolved issue?

esden, 2017-04-21:
As I am not getting a confirmation on this I assume this issue is resolved. If it is still an issue it should be reopened.

closed 2017-04-21 by esden