blackmagic-debug/blackmagic #104: STM32 F103 C8T6 undefined stack pointer

bitflipping, 2015-11-10:
The second revision of the STM32 F103 C8T6 placed on the STM32F411E-DISCO board seems to ignore the first word as stack pointer.
After erasing the option bytes, we got a "seg-fault" because the stack pointer was pointing to reserved memory (something like 0xFFFFFFe8).
It was manually set by gdb `set $msp=0x20005000` to be valid, so normal execution was possible.
The stack pointer is now 'floating' and will never be set to an initial value at startup or reset.
This problem could be solved by manually setting the $msp in the reset handler by opencm3, but the problem itself only shows by 'freeing' discovery boards, so the issue ended up here.
We advise manual patching the reset handler inside the stlink port of blackmagic probe.

gsmcmullin, 2015-11-11:
Is this true, even after a reset?

The gdb load command does not set the stack pointer, so if you attach to an erased device, and do 'load', then 'continue' or 'detach', a fault because of a garbage stack pointer is reasonable.  If you do 'load', then 'run' or 'kill', I would expect the target device to read the initial stack pointer from the flash, and work correctly.

bitflipping, 2015-11-11:
`kill` fixed it.
we removed the power source and thought this would achieve a hard reset.
seems not to work.
thanks alot

issue closed

closed 2015-11-11 by bitflipping