blackmagic-debug/blackmagic #105: new discovery boards support

huegene, 2015-11-12:
A few new discovery boards feature a highside switch to turn on or off the target µC. Default state is off, this way though the target reset line remains low all the time, preventing regular startup as the blackmagic_dfu assumes dfu boot mode. Since the PwrEn line is pulled up, its possible to detect its presence. 
quick fix:
diff new/usbdfu.c old/usbdfu.c 
38d37
<   uint16_t board_enable;
71,79c70
< 			GPIO_CNF_INPUT_PULL_UPDOWN, pin|GPIO15);
<   board_enable=gpio_get(GPIOB,GPIO15);/*if ST890 is presentbe will be 1*/
<   if(board_enable)
<   {
<     /*the ST890 is active low*/
<     gpio_set_mode(GPIOB,GPIO_MODE_OUTPUT_2_MHZ,
<         GPIO_CNF_OUTPUT_PUSHPULL,GPIO15);
<     gpio_clear(GPIOB,GPIO15);
<   }
---
> 			GPIO_CNF_INPUT_PULL_UPDOWN, pin);

huegene, 2015-11-12:
sorry, now a more appropriate patch  
[usbdfu.txt](https://github.com/blacksphere/blackmagic/files/33204/usbdfu.txt)

huegene, 2015-12-21:
Is there a propper way to add to the code base.  It just seemed like a very minor yet practical patch.

gsmcmullin, 2016-01-03:
Please create a pull request, and fix to use the Linux kernel coding style.  Then I will merge it.

gsmcmullin, 2016-07-12:
Applied in 0480578

closed 2016-07-12 by gsmcmullin