blackmagic-debug/blackmagic #111: srst still asserted after failed scan

lbonn, 2015-12-18:
When `connect_srst` is enabled, the pin stays asserted even if the scan failed. Then, it won't toggle on the next scans. This doesn't sound like the intended behaviour.

I tried to patch it in `jtag_scan` and `adiv5_swdp_scan` but it's not so straightforward because they can exit either normally or by an exception thrown in some subcalls.

The easiest way is to handle this in the caller code (`command.c`) but it's pretty ugly IMHO. Another way would be to isolate most of the scan routines code and do the assert/de-assert around it.

gsmcmullin, 2016-01-03:
The intention was that the reset signal is asserted on scan, and released on attach.  Do you need the reset released and reasserted on a failed scan?

lbonn, 2016-01-05:
Yes, I understand the intention and this mode is indeed quite handy but I'd prefer if it was released in case of a failed scan.

Usually I have this problem with bad contacts: I try a scan which fails and the device stays locked into reset while I can't do much with the debugger because I would need to scan again anyway.

It's mostly a minor annoyance but the basic idea is to return to the base idle state in case of failure.

PR #129 (2016-04-19): Cleanup of SRST handling.
- Remove connect_assert_srst global.
- Attach functions always release reset.
- Platforms provide a method to poll the reset pin.
- Reset on scan is all internal to command.c
- Reset is released on a failed scan.  Fixes #111
commit cd1d7225 (2016-04-19): Cleanup of SRST handling.

- Remove connect_assert_srst global.
- Attach functions always release reset.
- Platforms provide a method to poll the reset pin.
- Reset on scan is all internal to command.c
- Reset is released on a failed scan.  Fixes #111
closed 2016-04-19 by gsmcmullin