blackmagic-debug/blackmagic #113: stm32f7 write flash error

alex31, 2016-02-10:
I use to flash stm32f4 in swd or jtag mode with black magic probe without problem, but cannot do the same with stm32f7 even if logs suggests that F7 support exists.

Have tried in jtag or swd mode.  I am able to use all debug stuff under debugger, but sending a "load" command reports  "Error writing data to flash".

Is F7 support supposed to work with master, or is it a work in progress ?

BTW, flashing my F7 board works in DFU mode using dfu_util, or via stlink v2.1 probe and texane stlink. 

Alexandre

UweBonnes, 2016-02-10:
It wroks for me. Setup: F3 Discovery board as blackmagic debugger, connected to Nucleo-F7 via wire jumpers, one additional GND wire from board to board:

(gdb) mon erase_mass
Erasing flash... This may take a few seconds. /
(gdb) x /4x 0x08000000
0x8000000 <g_pfnVectors>:       0xffffffff      0xffffffff      0xffffffff      0xffffffff
(gdb) load
Loading section .text, size 0x10668 lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x8010668
Loading section .data, size 0xd94 lma 0x8010670
Start address 0x8000078, load size 70660
Transfer rate: 22 KB/sec, 954 bytes/write.
(gdb) x /4x 0x08000000
0x8000000 <g_pfnVectors>:       0x20003200      0x08000079      0x20000f49      0x20000f4d

alex31, 2016-02-12:
Happy to know that it should works.
My inhouse card use a F756 instead of a F746 on the nucleo board, should be the origin of my problem ?

when using stlink, it works : 

> 
/usr/local/bin/st-flash write build/ch.bin 0x08000000
2016-02-12T11:14:40 INFO src/stlink-common.c: Loading device parameters....
2016-02-12T11:14:40 INFO src/stlink-common.c: Device connected is: F7 device, id 0x10016449
2016-02-12T11:14:40 INFO src/stlink-common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
2016-02-12T11:14:40 INFO src/stlink-common.c: Attempting to write 10924 (0x2aac) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x8000
Flash page at addr: 0x08000000 erased
2016-02-12T11:14:40 INFO src/stlink-common.c: Finished erasing 1 pages of 32768 (0x8000) bytes
2016-02-12T11:14:40 INFO src/stlink-common.c: Starting Flash write for F2/F4/L4
2016-02-12T11:14:40 INFO src/stlink-common.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 10924
2016-02-12T11:14:40 INFO src/stlink-common.c: Starting verification of write complete
2016-02-12T11:14:41 INFO src/stlink-common.c: Flash written and verified! jolly good!

when using black magic probe, master version, sha ac9d87635c82112694991127de270c8abbcee7c6, compiled with launchpad 4.9-2015-q3-update release, it gives : 

> 
duke> arm-none-eabi-gdb build/ch.elf 
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F7xx
(gdb) attach 1
Attaching to program: /home/alex/DEV/STM32/CHIBIOS/C3DEV/DEVBOARDM7/RT-STM32F746G-DEVBM7/build/ch.elf, Remote target
_idle_thread (p=<error reading variable: can't compute CFA for this frame>) at ../../../ChibiOS_master/os/rt/src/chsys.c:75
75        }
(gdb) load
Loading section startup, size 0x200 lma 0x8000000
Error writing data to flash

After that, black magic is no more responding and need to be unplug/replug

Alexandre

alex31, 2016-02-13:
i have to add that, i can, with black magic probe, clear the eeprom via mon erase_mass with success, only load is failing.

Alexandre

mmoerdijk, 2016-02-13:
​Check the write protection settings of your micro. I have had some
problems with openocd and certain levels write protection. I think st link
clears the whole memory if you want to write to a readout protected chip
and then programs it.

you can check and set the write protection using st-link

alex31, 2016-02-24:
after some check, the problem is related with the NRST line. since NRST command in only needed in JTAG mode, and not in SWDIO mode, i just unwire it, and it now works. Just cannot understand why it worked fine with NRST wired with F4 and not with the F7

closed 2016-02-24 by alex31

UweBonnes, 2016-02-25:
To program the chips, we load a program to RAM that receives the data via JTAG and write the data to flash.  This requires reset to be released so the program can run, be it F7 or F4. I suspect that your F4 setup was different, actual releasing reset while you think it is asserted.