blackmagic-debug/blackmagic #122: Kinetis support may be corrupt

lyorak, 2016-04-08:
The current kinetis code returns flash erase and write commands with no error, so it appears to be correct. However in practice the erase and programming is just not performed due to a chain of bugs in the code. 
I have fixed it and the fix is available at https://github.com/lyorak/blackmagic/commit/c35deadbd4f56c8ee09784efc4c8d9a31b554abf
However I could try it only on Kinetis KL02, and do not want to initiate pull request for this change unless someone could try it in KL25 as well.
Changes: 
1. Fixed that the erase and program routines now check the return value of the command handler and report up the errors.
2. Fixed that one single programming failure will not make the programming impossible until reset.
3. Fixed the address offset of the Flash Common Command Object registers.

I believe that 1 and 2 can not cause problems on KL25  but am not 100% sure on point 3. However if you check the original code, it was using a byte swapping hack to address the last byte of the command register, due to the false assumption that writing 32 bits to this address would be correct. Instead, it is just writing 32bits consecutively from that address on, not backwards. I have changed this to point to the correct locations, and now load/file/comapre-sections gdb commands confirm that the file load actually has happened correctly on KL02. However  still happen that the KL25 is handling the addressing diffferently and the original (addr + (x)^3 ) code works there. 
Please suggest how to proceed.

dlaw, 2016-05-09:
I have confirmed KL25 flash memory programming works correctly with your changes to src/kinetis.c. I also added KL27 support based on your work, and will issue a pull request after I have had a few more days to test the changes. Note that I am working based on the tree at 8d9d6f6 because of https://github.com/blacksphere/blackmagic/issues/131.

dlaw, 2016-05-18:
Just issued a pull request including these changes:
https://github.com/blacksphere/blackmagic/pull/132

lyorak, 2016-05-19:
Great, thanks.

gsmcmullin, 2016-05-31:
Fixed by #132

closed 2016-05-31 by gsmcmullin