blackmagic-debug/blackmagic #128: GNU ARM 4.7.4 20130913 crash to build the firmware (requires GCC ARM 4.9.3)

bvernoux, 2016-04-18:
## Test1 KO GNU ARM Embedded  4.7.4 20130913
GNU ARM Embedded 4.7.4 20130913 tested on Windows 7 SP1 64bits (with 16GB RAM + SSD)
GNU_ARM_4_7_2013q3\bin\arm-none-eabi-gcc.exe --version
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 4.7.4 20130913 (re
lease) [ARM/embedded-4_7-branch revision 202601]

```
make
...
CC      main.c
cc1.exe: out of memory allocating 65536 bytes
make[1]: *** [main.o] Error 1
make: *** [all] Error 2
```

## Test2 OK ARM Embedded 4.9.3 20150529 OK
GNU ARM Embedded  4.7.4 20130913 tested on Windows 7 SP1 64bits (with 16GB RAM + SSD)
GNU_ARM_4_9_2015q3\bin\arm-none-eabi-gcc.exe --version
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]

```
make
...
CC      main.c
LD      blackmagic
OBJCOPY blackmagic.bin
LD      blackmagic_dfu
OBJCOPY blackmagic_dfu.bin
OBJCOPY blackmagic_dfu.hex
```

bvernoux, 2016-04-18:
Note I have not tested on GNU_ARM_4_8_2014q3, GNU_ARM_4_9_2014q4, GNU_ARM_4_9_2015q2 ... so it probably works on those version
It is just a big warning for anyone using GNU ARM 4.7.4 20130913

gsmcmullin, 2016-10-21:
This is not a bug in the blackmagic firmware.

closed 2016-10-21 by gsmcmullin