blackmagic-debug/blackmagic #131: 459bae4 breaks swdp_scan

dlaw, 2016-05-09:
I'm using the Black Magic Probe to debug a custom Kinetis KL27 board. All was well until I upgraded the firmware of my probe. I bisected the commit history and found that 8d9d6f6 works fine, but under 459bae4 "mon swdp_scan" fails with the following output:
```
Target voltage: 3.3V
SW-DP scan failed!
```

I switched between the two BMP firmware revisions several times and was careful to perform exactly the same sequence of steps each time:
1. attach BMP by USB
2. power up target device
3. attach SWD cable between probe and device
4. launch GDB and execute "target extended-remote /dev/ttyACM0" folllowed by "mon swdp_scan"

Happy to perform any additional tests that will provide useful information. Cheers, David

gsmcmullin, 2016-05-09:
The problem here is the updates to the ADIv5 code to probe the ROM table to identify the debug components.  This broken for all Cortex-M0/M0+ devices that are no longer identified.  PR #130 attempts to fix this by always calling `cortexm_probe` which isn't ideal.  I will look into the ROM table structure of a Cortex-M0 device and see if I can provide a better fix.,

OpenUAS, 2016-05-29:
"...provide a better fix..."
@gsmcmullin Thx, that would really be appreciated.

PR #133 (2016-05-31): adiv5: Probe ARMv6M devices with Cortex-M driver.
Working on LPC812.
Fixes #131. Supersedes #130.
closed 2016-05-31 by gsmcmullin