blackmagic-debug/blackmagic #145: Last firmware breaks on STM32F0 devices

adamgreig, 2016-07-24:
Running 5832d8a42a16b862462d7cd91f2b986198a08ed5, I can `mon swdp_scan` and see my STM32F05x just fine, as is usually the case. I can't build the next commit b03cc59bea43ce93b15262d986caf3aecc03cca5 (build errors) but by the following commit b64472452075432d08ac214e1b46bbc25dbd23d4 attempting to swdp_scan instead yields:

`Target voltage: 3.3V
Remote communication error.  Target disconnected.: Input/output error.`

This is a stock BMP mini (v1) attached to a very normal STM32F05. Everything continues to work as normal for an F1 I have lying around, but I haven't tested on any other devices.

gsmcmullin, 2016-07-25:
Hi Adam

Thank you for the bug report.  I don't see anything in the commits you mention that will affect the swdp_scan, and the STM32F0 and STM32F1 use the exact same code path.  I'll search for a board with an STM32F05 and see if I can reproduce the problem.

Also, from the error it looks like the BMP was disconnected on USB.  The code to disconnect the USB is only called if USB VBUS loses power.  Can you please confirm that there is no problem with your USB cable.

adamgreig, 2016-07-28:
I agree it's pretty weird.

On a different STM32F05 target, with a new USB cable, same BMP mini r1:

5832d8a works fine, ten times in a row

b644724 fails every time, ten times in a row, with the error I pasted above, and a USB disconnect in dmesg accompanying each failure (so I have to unplug and replug it each time)

master (f1314601) has the same behaviour as b644724.

My guess is a crash in the firmware causing USB to disassociate? I've not really looked into it much and have misplaced my picoblade -> JTAG cable so can't easily hook the BMP up to another debugger for at least a few days.

commit ca364a88 (2016-08-21): Fix #145

Don't tc_printf in the probe.
tc_printf fixed to not crash of no controller connected.
closed 2016-08-21 by gsmcmullin