blackmagic-debug/blackmagic #148: Can't find BMC target under gdb OSX Capitan

pstrupler, 2016-08-18:
Hi all,

Since upgrading OSX from yosemite to capitan I'm not able to flash anymore.

I can see the device under /dev/tty.usbmodemXXXXXX but when I'm targeting it with target extended-remote /dev/tty.usbmodemXXXXXX in gdb nothing happens - not even an error.

Flashing using my Ubuntu virtual machine on OSX Capitan still works though.

I've got the Blackmagic V2 with the current 1.6 release flashed.

gsmcmullin, 2016-08-18:
Does it work with the /dev/cu.usbmodem* device?  The tty device can block
forever on open if DCD isn't asserted.

esden, 2016-08-18:
I have just tested on Mac OS X "10.11.6". It works perfectly fine with /dev/cu.usbmodemXXXXXXX1 and blocks on /dev/tty.usbmodemXXXXXXX1. But that has always been the case as far as I remember. I am surprised you got the tty device to work on yosemite.

@pstrupler where have you find a reference to using tty device on OS X? If we have it somewhere in our documentation this should be corrected.

esden, 2017-02-23:
I will close this issue as it seems to be user specific. Feel free to reopen this issue if it is still present and needs addressing.

closed 2017-02-23 by esden