blackmagic-debug/blackmagic #149: Issues loading code to STM32L475xG

russss, 2016-08-30:
I'm trying to load code on to an STM32L475 (the [EMF TiLDA Mk3 badge](https://badge.emfcamp.org/wiki/TiLDA_MK3)). Debugging seems to work fine, but the "load" command never succeeds. It seems to happen with any .elf file I try and throw at it:

```
(gdb) load
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x6fd90 lma 0x8020000
Error writing data to flash
```

```
(gdb) load
Loading section vectors, size 0x1a0 lma 0x8000000
Loading section .text, size 0x5e94 lma 0x80001a0
Loading section .rodata, size 0x1320 lma 0x8006034
Error writing data to flash
```

In addition, `monitor erase_mass` seems to leave the chip in an unusable state:

```
(gdb) monitor erase_mass
Erasing flash... This may take a few seconds.  
(gdb) load
Error erasing flash with vFlashErase packet
```

This is reproducible with every BMP I have managed to get my hands on, both with firmware version v1.6-rc0-121-g4d6f691 and git revision 45e322b9a332c41930d3a781830805ecd27e1682 (HEAD currently).

Perhaps this is related to [this comment](https://github.com/blacksphere/blackmagic/pull/119#issuecomment-192600675) on #119?

chmousset, 2016-09-29:
Hello there,

First of all thanks for the marvellous tool BMP is! We use plenty of them at the office and are happy about it :)
This is so much easier than using st-link!

I'm having exactly the same issue as russss, but with STM32L476RGT6U on NUCLEO board, both with on-board debug MCU (reflashed as BMP), and with an external BMP.
Both BMP version v1.6-rc0-215-g45e322b

Additionnaly, erasing does not work. memory is unmodified after using both erase_mass and erase_bank1.

Has the STM32L4 support been tested with actual boards and FW? Which ones?

gsmcmullin, 2016-09-30:
The STM32L4 driver was contributed by @UweBonnes.  Hopefully he can answer your questions.
I don't have any STM32L4 hardware and haven't tested this driver at all myself.

UweBonnes, 2016-09-30:
I will try to have a look this weekend.

chmousset, 2016-09-30:
Hey guys, thanks for the answer.

I Got further into the issue: In STM32L4xx writes should be 8-bytes aligned as per RM0351, but as in ChibiOS the sections are setup as 4-bytes aligned in the linker file some write operations may be 4-bytes aligned.

Somehow BMP extends these 4-bytes aligned to 8-bytes align (target_flash_write() in target.c ?), possibly overwriting the FLASH memory.

Ex:
```
Loading section vectors, size 0x1a0 lma 0x8000000
Loading section .text, size 0x5ef4 lma 0x80001a0
Loading section .rodata, size 0x1370 lma 0x8006094 (this tells the stub to write at 0X08006090, overwriting the last section)
Error writing data to flash
```

I modified my linker file to align on 8 bytes and writes are now OK!

Now, is there a cleaner way to do this?
 * Modify the linker to put all the FLASH sections into a new single big one (therefore only one write command)
 * have BMP concatenate sections not 8-bytes aligned

UweBonnes, 2016-09-30:
As there is no way to cope with a half written double word, only forcing  8byte alignment makes sense. Is there perhaps a better way for error output? The output of the failing "load" command above is not helpfull!

gsmcmullin, 2016-09-30:
Using the buffered flash model to aggregate writes into complete sectors should solve this.
Try this (totally untested): https://github.com/blacksphere/blackmagic/compare/master...gsmcmullin:stm32l4_buffered

gsmcmullin, 2016-10-26:
Did anyone try this?  Does it solve the problem?  @UweBonnes @chmousset

UweBonnes, 2016-10-26:
As told before, only 8 byte can be written at once and no less. So this is not a bug.  The original poster was given something to try out, but he never got back...

russss, 2016-10-26:
The linker file hack works for me. I'm unable to flash new firmware onto my blackmagic clone so I can't test @gsmcmullin's branch.

gsmcmullin, 2016-10-26:
@UweBonnes It is a problem if an 8 byte block is covered by 2 different sections.  GDB will send this out in two separate write packets, and the second will fail.  There is no reason why they cant be consolidated into a single write in the BMP firmware, that is what my branch is intended to solve, but I don't have hardware to test it on.

UweBonnes, 2016-10-27:
As the L4 has ample flash, 8-byte alignment is probably the best solution, as "caching"is also done in 8-byte units. Despite that, I will try to adapt BM for the Stlink V2-1 on the L476 disco, but also try to test the buffered flash model.  L4 has direct access to the Stlink SWD pins, so no change n the SB5/6/9/and 10 solder jumpers is needed.

UweBonnes, 2016-10-31:
Yes, your patch mentioned above helps in the situation when a section does not start  8-byte aligned.

Please apply!

gsmcmullin, 2016-10-31:
Merged.  Thank you for testing.

PR #167 (2016-10-31): STM32L4: Use buffered flash model.
Fixes #149
closed 2016-10-31 by gsmcmullin