blackmagic-debug/blackmagic #155: New nRF51822 device code needs to be added

rogerclarkmelbourne, 2016-10-06:
I have just come across a new nRF51822 device IDE that does not appear in nrf51.c

On a new batch of boards I was programming one device was not recognised, and I have now checked its code and markings.

The device is marked as nRF51822QFAAH1 and its device ID is showing as 0x008F

I will post to the Nordic forum to confirm this, as Nordic don't seem to maintain a good accurate list of device codes, and update this issue when I get a response from Nordic

rogerclarkmelbourne, 2016-10-06:
To update this issue.

I had a response for a Nordic employee via the Nordic developer forum, and he confirmed the 0x008F is the correct ID for the nRF51822QFAAH1

See 

https://devzone.nordicsemi.com/question/97769/can-someone-conform-the-config-id-code-for-the-nrf51822qfaah1/?answer=97824#post-id-97824

gsmcmullin, 2016-10-09:
Hi Roger.  Thank you.  Can you please make the update, confirm that it works for you and then send a pull request?  I don't have any Nordic devices for testing.

rogerclarkmelbourne, 2016-10-09:
Thanks Gareth.

I only have one of these devices, in a batch of 50 boards from a supplier in China. 

But I have now had confirmation from Nordic, in response to my question, so I'm certain that the new device code is 0x008F

I will do a PR, but it won't be for a few days, because my existing fork of your repo is considerably out of date and has other additions, which are nothing do to with this simple change.

So I'll need to do some housekeeping first ;-)

rogerclarkmelbourne, 2016-10-16:
See
https://github.com/blacksphere/blackmagic/pull/158

closed 2016-10-16 by gsmcmullin