blackmagic-debug/blackmagic #156: Unable to read nRF51 device FICR registers in gdb

rogerclarkmelbourne, 2016-10-06:
It would be good if gdb could read more of the nRF51 "Factory Information Configuration Registers (FICR)"

Currently only this block of registers can be accessed (read or written)

	nrf51_add_flash(t, NRF51_UICR, 0x100, 0x100);

Probably the easiest way to do this is to add this line

	nrf51_add_flash(t, NRF51_FICR, 0x100, 0x100);

Below the UICR registers line

But perhaps using nrf51_add_flash is not the best way, and I should use nrf51_add_ram (even though its not RAM), to stop the flash erase code treating these as flash ???

gsmcmullin, 2016-10-09:
A better way to do it is to use the command `set mem inaccessible-by-default-off` from GDB (or add to your `.gdbinit`).  This allows GDB to access memory outside of the memory map as RAM.

rogerclarkmelbourne, 2016-10-09:
Thanks Gareth

I'll give that a try. (and not add those reg's as flash.

closed 2016-10-16 by rogerclarkmelbourne