blackmagic-debug/blackmagic #159: Would it be possible to add a binaries folder to save people having to compile for their target

rogerclarkmelbourne, 2016-10-16:
It think it would be worthwhile adding a binaries folder with precompiled versions, as currently anyone who wants to flash the Blackmagic probe to their hardware needs to clone the whole repo, update the submodules, and Make before they can actually program their hardware.

I'm happy to build for all the ARM targets using gcc 4.9.3 and to submit a PR with this.

gsmcmullin, 2016-10-21:
Hi Roger

I don't think they belong in the repo, but I agree they should be available.  I used to have Jenkins build them and upload them to a webserver, but I decomissioned that a while ago.  I now have automatic builds on Travis, but the binaries aren't archived anywhere.  I agree we should do this and I'll look at it when I have a chance.

rogerclarkmelbourne, 2016-10-21:
Hi Gareth

No worries.

I know strictly speaking that git is for source control, but I know Arduino have compiled firmware in their repo https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/firmwares/atmega16u2/Arduino-DUE-usbserial-prod-firmware-2012-11-05.hex

But as long as people have somewhere to download them from , I don't suppose it matters where it is, as long as they can find it ;-)

gsmcmullin, 2016-10-22:
After a lot of messing around with Travis, I finally have this sort of working.
The latest build will be available here: http://builds.blacksphere.co.nz/blackmagic

rogerclarkmelbourne, 2016-10-22:
Thanks.

I guess this link now needs to be added to the readme ;-)

gsmcmullin, 2016-10-22:
Agreed and done.

closed 2016-10-22 by gsmcmullin

rogerclarkmelbourne, 2016-12-06:
@gsmcmullin 

I tried to flash the stlink binary but it didnt work.

But discussing what's going wrong with @rickkimball, I realised that the bin file is just the main code segment i.e blackmagic.bin and doesnt seem to contain the dfu segment (blackmagic_dfu.bin)

So the binaries seem to be unless as they are missing a fundamental part.

I think the binaries need to contain both the dfu and the main code segments, otherwise its not possible to flash a target system, (as you still need to compile your own dfu segment)

gsmcmullin, 2016-12-06:
These binaries work for firmware upgrading, according to our instructions:
https://github.com/blacksphere/blackmagic/wiki/Upgrading-Firmware

If you have a Black Magic Probe, it includes the DFU bootloader so there is no need to compile it.

rogerclarkmelbourne, 2016-12-06:
OK.

You may as well remove the stlink binary, as its not possible to use DFU to update the STM Nucleo or Discovery boards, as the existing firmware on those boards uses its own special encrypted protocol for upgrades (and STM have not published the details of how to generate the encrypted binaries or write your own updater)

I'm not sure about the F4 discovery, I presume that its basically also just a STLink in the first place, and there will be no way to "upgrade" without first flashing the F4 specific DFU binary

gsmcmullin, 2016-12-06:
Once you have 'liberated' your st-link, and replaced the ST firmware with the blackmagic DFU bootloader, you should be able to update the firmware over DFU.  We have no objections to people reflashing cheap hardware with the blackmagic firmware, but the details of doing so are beyond scope of the project (unless you're using an official Black Magic Probe to reflash your ST-link)

rogerclarkmelbourne, 2016-12-06:
No worries.

People have already posted the combined binaries elsewhere.