blackmagic-debug/blackmagic #162: LEDs 0-1-2 stops blinking

imuguruza, 2016-10-20:
Hi there,

I'm trying to reflash a STM32F1xx chip but I think I'm losing connection...
In `gdb`, when trying to attach it gets blocked. And time to time next error prints:
`Ignoring packet error, continuing...`

Hardware wise, only power LED is ON, the LEDs 0 to 2 are always `off`. 
Any idea what coulb be wong?

Thanks!

gsmcmullin, 2016-10-20:
This is certainly a bug.  It appears to be getting stuck in an endless loop somewhere.
Can you please try again, but issue the command `set debug remote 1` before connecting, the will output all communicatoin with BMP.  If you can send me the log, I'll take a look and seee if I can find where the problem is.

imuguruza, 2016-10-20:
Great, I'll do that asap.

imuguruza, 2016-10-21:
Here some captures:
- When `target extended-remote /dev/ttyACM0` , it prints [this](https://gist.github.com/imuguruza/118c0492b005dd9fddf8a87ef894b3f8)
- mon swdp_scan [gist](https://gist.github.com/imuguruza/c655ec245fd6fb96b126b12db4287096)
- `attach 1` [gist](https://gist.github.com/imuguruza/566c1920f11d74cdc51f9b2d58f06e43)

Gets freezed there, no more prints at all. 

Even pressing `ctrl`+ `c` does not exit, I have to unplug the USB cable.

If you explain me how to create a log, I'll do it

gsmcmullin, 2016-10-21:
Thanks.  To capture a log from gdb, do `set logging on`.  It will write all output to a file `gdb.txt`.

From you log above, everything is quite normal until you give the `attach` command.  I've just taken a look through the code and can't see any way that it could get stuck.  Can you please confirm you are using the latest firmware?  You can do `mon ver` to find out what it is currently programmed with.

rogerclarkmelbourne, 2016-10-21:
Try

set pagination off
set logging file gdb_log.txt
set logging on

imuguruza, 2016-10-21:
I'm using v1.6-rc0
```
(gdb) mon ver
Black Magic Probe (Firmware v1.6-rc0)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
```

imuguruza, 2016-10-25:
Just to let you know, I have reflashed a stucked board using `stlink`(st-util).

gsmcmullin, 2016-10-25:
Sorry for not getting back to you.  That v1.6-rc0 firmware is very old, you should try with the current master.  Are you able to attach now that the target board has been reflashed?

imuguruza, 2016-10-25:
> Sorry for not getting back to you
No worries.

I have been working with `bmp` once I managed to come back to life the micro.
So, should I update to `master` ?

gsmcmullin, 2016-10-25:
Ok, thanks.  So the problem was related to whatever was in the flash at the time.  Do you know what that was?  If it was erased, this could be #5 that I can't reproduce (and I've flashed hundreds of boards).

If v1.6-rc0 is working for you there is no need to update, but if you run into issues, try update to master as it's possible that they've already been fixed.  Also bug reports against master will be more useful.

imuguruza, 2016-10-26:
Well, the problem was at running time. Once running got unresponsive. Pressing `ctr` + `c` at gdb could not stop the execution. So I had to abort it. After, I was not able to `attach 1` the board. The output is on [gist](https://gist.github.com/imuguruza/566c1920f11d74cdc51f9b2d58f06e43)

esden, 2017-02-24:
Is this still an issue? Would attaching a secondary BMPM to the primary BMPM help to see where the code get's stuck and why?

imuguruza, 2017-02-24:
Nevermind, right know I'm not using it... I don't have two BMP

closed 2017-02-24 by imuguruza

esden, 2017-02-24:
Thank you for following up. And if you run into this issue again make sure to let us know so we can see if there is a way to prevent that. :)