blackmagic-debug/blackmagic #172: Enhancement - Do you think it would be practical to add USB mass storage as an upload method?

rogerclarkmelbourne, 2016-11-25:
Do you think it would be practical to add USB mass storage as an upload method, to operate like the CMSIS DAP and some STLink adaptors do, to flash the code onto the target.

I've not looked at the code, but I presume its a composite USB device ? so an additional device could be added.

I was thinking that the code to upload could use monitor swdp_scan followed by attach 1 to attach to whatever the first device was, and then load the bin etc as if the commands came from GDB.

However there may be some technical gotya's that I'm not aware of e.g problem with the USB composite device ?

gsmcmullin, 2016-11-25:
It is a composite USB device, but we do not have enough free endpoints to add MSC support.

For me, at least, it's use as a debugger is far more valuable than a dumb programmer.  Using a GDB script to use it as a dumb programmer is also not too much effort.

esden, 2016-11-25:
I think that adding the mass storage flashing will in a sense hide the debugging capability from the user. If you are able to compile your code you will very likely have gdb already on your computer. So flashing should be as easy as:
`arm-none-eabi-gdb --batch -ex 'tar ext /dev/ttyACM0' flash.scr binary.elf`

With the following script as your `flash.scr`:
```
monitor swdp_scan
attach 1
load
kill
```

Or alternatively a more fancy version:
```
monitor version
monitor swdp_scan
attach 1
load
compare-sections
kill
```

If you ask me this is also easier to integrate with your build system than having to know what disk mount is the actual target. (the handling of disk mounting varies a lot between windows, mac os or linux and in linux it differs even between distributions) using gdb is likely to be much easier to make consistent between platforms.

rogerclarkmelbourne, 2016-11-25:
I already use the BMP to flash using a GDB via the command line like this

`"{path}{cmd}" -cd "{build.path}" -b {upload.speed} {upload.verbose} -ex "set debug remote 0" -ex "set target-async off" -ex "set remotetimeout 60" -ex "set mem inaccessible-by-default off" -ex "set confirm off" -ex "set height 0" -ex "target extended-remote {serial.port}" -ex "monitor swdp_scan" -ex "attach 1" -ex "x/wx 0x8000004" -ex "monitor erase_mass" -ex "echo 0x8000004 expect 0xffffffff after erase\n" -ex "x/wx 0x8000004" -ex "file {build.project_name}.elf" -ex "load" -ex "x/wx 0x08000004" -ex "tbreak main" -ex "run" -ex "echo \n\n\nUpload finished!" -ex "quit"`

But driver issues are always a problem.

People expect plug and play

@gsmcmullin 
Can't I just increase the number of endpoints in my version ?

Note. I appreciate you don't want to add this to your master version, so I will need to make sure whatever way I do it, that I can keep my fork up to date by still being able to pull your master version

gsmcmullin, 2016-11-25:
@rogerclarkmelbourne The short answer is no.  I can't remember if the real limitation is the number of endpoints supported in hardware, or the amount of USB packet memory, but I'm pretty sure you can't just add more bulk endpoints needed to implement USB MSC. You're welcome to try though, I'll be pleased to be proved wrong.

If the limitation is packet memory you could reduce the max packet size on the existing endpoints to free up packet memory, this will come at some cost of throughput reduction though.

gsmcmullin, 2016-11-26:
I'm also not sure what you plan to implement.  If this is a block device, so you can `dd if=blackmagic.bin of=/dev/sdb`, even that is a bother on Windows.  If this is a fake filesystem so you can `cp blackmagic.hex /media/bmp/`, then you're really opening up a can of worms.

rogerclarkmelbourne, 2016-11-26:
I was going to try the fake file system route. I had some code for this on the STM32, but it sounds more complex to add than I'd anticipated.

rogerclarkmelbourne, 2016-11-27:
@gsmcmullin  
We may as well close this, and I'll re-open if I get time to investigate this option, but I may do it a different way entirely, as I can build code for the STM32 with a USB mass storage + FS from STM's on boilerplate code, so it may be easier for me to patch in your SWD code into that, rather than trying to do it the other way around albeit at the expense of loosing the debugging side of things (which I hoped to keep)

gsmcmullin, 2016-11-27:
I'll close it then.  Unless things have changed, the USB code here is much easier to understand than what ST had.  I'm too busy with other things at the moment to take a look at this now.  Let us know if do get something working.

closed 2016-11-27 by gsmcmullin