blackmagic-debug/blackmagic #177: AVR target support

gsmcmullin, 2016-12-13:
This is not actively been worked on.  This issue is just a placeholder for reference to relevant documentation and/or other references.

gsmcmullin, 2016-12-13:
AVR OCD registers are not publicly documented, but have been partially reverse engineered.

See unofficial documentation here:
https://people.ece.cornell.edu/land/courses/ece4760/FinalProjects/s2009/jgs33_rrw32/Final%20Paper/Documentation.html

aster94, 2018-08-28:
I am commenting just to express the desire that the BMP one day (_hope soon_) will support the AVR ecosystem

Also, maybe since the opening of the issue in 2016 there are new possibilities/opportinities to see this achieved

5N44P, 2018-08-28:
I also want to express all my intrest towards this feature. 
For me, the BMP would become the definitive debugger. 

If i'm not mistaken, the old JTAGICE mkII specification is released by Atmel itself (http://www.microchip.com//wwwAppNotes/AppNotes.aspx?appnote=en591238)

And also there's this: http://www.ruemohr.org/docs/debugwire.html

Testato, 2018-08-28:
the dW support is very important because it is used on classica 328P (Arduino Uno) so many people may adoptthe BMP instead of buy the Atmel-ICE

This is an open implementation
https://github.com/dcwbrown/dwire-debug

5N44P, 2018-09-02:
> BMP is an open source effort. Why not try to add the support yourself? My main effort is also only programming STM32, so my BMP contributions for STM32 were also only a side effect.

I'd try to, but that's WAY beyond my skills. I'm an hardware designer, i do little to nothign software...

UweBonnes, 2020-12-31:
https://github.com/UweBonnes/blackmagic/tree/avr as a start can program AT90CANxxx. JTAG ID for other devices and more things need to be added.

DX-MON, 2022-03-16:
This issue should be closed (or at least partially addressed) by #992 when we complete that work and get it to land. A complete documentation of the JTAG-PDI system has been undertaken. The limiting factor is parts and a little guesswork as to how exactly certain parts of the debug machinery work which has limited the speed of progress