blackmagic-debug/blackmagic #178: Legacy ARM target support

gsmcmullin, 2016-12-13:
This is not actively been worked on. This issue is just a placeholder for reference to relevant documentation and/or other references.

gsmcmullin, 2016-12-13:
Incomplete ARM7TDMI driver removed from current firmare:
https://github.com/blacksphere/blackmagic/blob/ee3af96a736139b71678a0e590865f9ffafc2f35/src/arm7tdmi.c

For ARM11 (Raspberry Pi 1) support see chapters 13 and 14 from:
http://infocenter.arm.com/help/topic/com.arm.doc.ddi0301h/DDI0301H_arm1176jzfs_r0p7_trm.pdf

esden, 2021-12-15:
As this is marked as Won't fix I am closing this issue. In case someone decides to work on adding support for old ARM targets this still can be referred to.

closed 2021-12-15 by esden