blackmagic-debug/blackmagic #179: Error on running `make` in root dir

paramaggarwal, 2017-01-01:
```
target/adiv5.c: In function 'adiv5_dp_init':
target/adiv5.c:391:11: error: variable 'ctrlstat' might be clobbered by 'longjmp' or 'vfork' [-Werror=clobbered]
  uint32_t ctrlstat = 0;
           ^~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [adiv5.o] Error 1
make: *** [all] Error 2
```

My toolchain (macOS Sierra):
```
$ arm-none-eabi-gcc -v
gcc version 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739] (GNU Tools for ARM Embedded Processors) 
```

Complete logs:
```
$ make clean
  CLNHDR  include/libopencm3/efm32/efm32g/irq.json
  CLNHDR  include/libopencm3/efm32/efm32gg/irq.json
  CLNHDR  include/libopencm3/efm32/efm32lg/irq.json
  CLNHDR  include/libopencm3/efm32/efm32tg/irq.json
  CLNHDR  include/libopencm3/lm3s/irq.json
  CLNHDR  include/libopencm3/lpc13xx/irq.json
  CLNHDR  include/libopencm3/lpc17xx/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m0/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m4/irq.json
  CLNHDR  include/libopencm3/sam/3a/irq.json
  CLNHDR  include/libopencm3/sam/3n/irq.json
  CLNHDR  include/libopencm3/sam/3s/irq.json
  CLNHDR  include/libopencm3/sam/3u/irq.json
  CLNHDR  include/libopencm3/sam/3x/irq.json
  CLNHDR  include/libopencm3/stm32/f0/irq.json
  CLNHDR  include/libopencm3/stm32/f1/irq.json
  CLNHDR  include/libopencm3/stm32/f2/irq.json
  CLNHDR  include/libopencm3/stm32/f3/irq.json
  CLNHDR  include/libopencm3/stm32/f4/irq.json
  CLNHDR  include/libopencm3/stm32/l1/irq.json
  CLEAN   lib/stm32/f0
  CLEAN   lib/stm32/f1
  CLEAN   lib/stm32/f2
  CLEAN   lib/stm32/f3
  CLEAN   lib/stm32/f4
  CLEAN   lib/stm32/l1
  CLEAN   lib/lpc13xx
  CLEAN   lib/lpc17xx
  CLEAN   lib/lpc43xx/m4
  CLEAN   lib/lpc43xx/m0
  CLEAN   lib/lm3s
  CLEAN   lib/lm4f
  CLEAN   lib/efm32/efm32tg
  CLEAN   lib/efm32/efm32g
  CLEAN   lib/efm32/efm32lg
  CLEAN   lib/efm32/efm32gg
  CLEAN   lib/sam/3a
  CLEAN   lib/sam/3n
  CLEAN   lib/sam/3s
  CLEAN   lib/sam/3u
  CLEAN   lib/sam/3x
  CLEAN   doc
  CLEAN
$ make
  GENHDR  include/libopencm3/efm32/efm32g/irq.json
  GENHDR  include/libopencm3/efm32/efm32gg/irq.json
  GENHDR  include/libopencm3/efm32/efm32lg/irq.json
  GENHDR  include/libopencm3/efm32/efm32tg/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
[... 438 lines cut ...]
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_sam3x.a
  GIT     include/version.h
  CC      target/adiv5.c
target/adiv5.c: In function 'adiv5_dp_init':
target/adiv5.c:391:11: error: variable 'ctrlstat' might be clobbered by 'longjmp' or 'vfork' [-Werror=clobbered]
  uint32_t ctrlstat = 0;
           ^~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [adiv5.o] Error 1
make: *** [all] Error 2
$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/Caskroom/gcc-arm-embedded/6_2-2016q4,20161216/gcc-arm-none-eabi-6_2-2016q4/bin/../lib/gcc/arm-none-eabi/6.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: /tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/src/gcc/configure --target=arm-none-eabi --prefix=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/install-native --libexecdir=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/install-native/lib --infodir=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/install-native/arm-none-eabi --build=x86_64-apple-darwin10 --host=x86_64-apple-darwin10 --with-gmp=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/build-native/host-libs/usr --with-mpfr=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/build-native/host-libs/usr --with-mpc=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/build-native/host-libs/usr --with-isl=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/build-native/host-libs/usr --with-libelf=/tmp/jenkins-GCC-6-build_build-toolchain-mac-93_20161216_1481890191/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=rmprofile
Thread model: single
gcc version 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739] (GNU Tools for ARM Embedded Processors) 
```

fcgdam, 2017-01-05:
I'm also having this issue on Linux and with the latest ARM toolchain version. 

The error is the same.

paramaggarwal, 2017-01-05:
@fcgdam You can fix this meanwhile by removing the `Werror` parameter in this line: https://github.com/blacksphere/blackmagic/blob/master/src/Makefile#L13

fcgdam, 2017-01-05:
Confirmed. It works now and outputs the firmware. Thanks.

paramaggarwal, 2017-01-05:
I have documented some steps here: https://www.hackster.io/paramaggarwal/converting-an-stm32f103-board-to-a-black-magic-probe-e701d4?ref=user&ref_id=127978&offset=1

closed 2017-01-08 by gsmcmullin

paramaggarwal, 2017-01-09:
Thanks @gsmcmullin.

PR #183 (2017-01-08): adiv5_dp_init: make ctrlstat volatile to supress warning.
Fixes #179