blackmagic-debug/blackmagic #181: load binary fail with gdb 7.12 when size > 128k

alex31, 2017-01-05:
I meet a problem with arm-none-eabi-gdb. With gdb 7.10 (which come with gcc5.x), all is fine, but
with gdb 7.12  (which come with gcc6.x), load fail only if .bin is more than 128ko (target is an stm32f4)

trace with gdb 7.10 (when it works) :

philcrump, 2017-01-26:
I'm also seeing this.

> $ arm-none-eabi-gdb --version
> GNU gdb (GNU Tools for ARM Embedded Processors) 7.12.0.20161204-git

Target: STM32F429ZI (Ethernet Nucleo)

UweBonnes, 2017-02-13:
With 7.12 erasing larger chunks also times out for me

gsmcmullin, 2017-02-20:
The problem appears to have been introduced in GDB, in commit 048094acc.

Function `wait_for` in `ser-unix.c` is probably the source of the issue.  It was rewritten to timeout using `select` instead of setting `VTIME`/`VMIN`.

UweBonnes, 2017-02-20:
How to proceed? And why does does gdb not honor setting a larger timeout?

gsmcmullin, 2017-02-20:
GDB bug report: https://sourceware.org/bugzilla/show_bug.cgi?id=21188

I'll try send in a patch for GDB.  Even if it's accepted, it'll take a long time before it finds its way downstream to gcc-arm-embedded.

esden, 2017-03-21:
The patch for this has been merged upstream: https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=commit;h=9bcbdca808b5f9fec6217d20bd4b48a56008c460 I assume this will become available with the 8.0 release that is being prepared at the moment.

gsmcmullin, 2017-06-05:
GDB 8.0 has been released, which includes this fix.

UweBonnes, 2021-12-15:
The reference to this issue in #89 was not meant to reopen #181!

closed 2021-12-15 by UweBonnes