blackmagic-debug/blackmagic #185: Unable to flash blackmagic firmware to Launchpad TM4C123GXL ICDI

hazelnusse, 2017-01-30:
I have the following setup:
- BMP2 running the blackmagic firmware (sha: 7ba2987a83bab1adb1db9b7f167e2454c7cb9b44)
- EK-TM4C123GXL, power switch set to DEBUG
- Tag Connect cable: TC2050-IDC-NL

I first connect the BMP2 to my laptop via the USB cable, here is what I see from dmesg:
[97703.515430] usb 2-3: New USB device found, idVendor=1d50, idProduct=6018
[97703.515433] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[97703.515435] usb 2-3: Product: Black Magic Probe
[97703.515436] usb 2-3: Manufacturer: Black Sphere Technologies
[97703.515437] usb 2-3: SerialNumber: D5DABEF4
[97703.516025] cdc_acm 2-3:1.0: ttyACM0: USB ACM device
[97703.516387] cdc_acm 2-3:1.2: ttyACM1: USB ACM device

So far so good.

Next, I connect the EK-TM4C123GXL to my laptop via the debug USB, here is what I see from dmesg:
[97807.165383] usb 2-1: new full-speed USB device number 66 using xhci_hcd
[97807.334651] usb 2-1: New USB device found, idVendor=1cbe, idProduct=00fd
[97807.334654] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[97807.334656] usb 2-1: Product: In-Circuit Debug Interface
[97807.334657] usb 2-1: Manufacturer: Texas Instruments
[97807.334658] usb 2-1: SerialNumber: 0E210604
[97807.335134] cdc_acm 2-1:1.0: ttyACM2: USB ACM device

And the LEDs are blinking on the EK-TM4C123GXL.

Next, I connect the Tag Connect to the BMP2 hardware via the 10pin connector. Next, I connect the other end to the Tag Connect JTAG connection on the EK-TM4C123GXL.

Then, I build the blackmagic firmware for the launchpad-icdi by running:
$ make PROBE_HOST=launchpad-icdi

This results in these two files:
```
ls -l src/blackmagic*
-rwxr-xr-x 1 luke luke 894312 Jan 29 21:55 src/blackmagic*
-rwxr-xr-x 1 luke luke  71968 Jan 29 21:55 src/blackmagic.bin*
```
Now, I run gdb, connect, and try to load the program:
```
arm-none-eabi-gdb ./src/blackmagic
GNU gdb (GNU Tools for ARM Embedded Processors) 7.12.0.20161204-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./src/blackmagic...done.
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor jtag_scan
Target voltage: 1.8V
Available Targets:
No. Att Driver
 1      TI Stellaris/Tiva
(gdb) monitor tpwr enable
(gdb) monitor jtag_scan                                                                                                                                                                                                                       
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      TI Stellaris/Tiva
(gdb) attach 1
Attaching to program: /home/luke/repos/blackmagic/src/blackmagic, Remote target
kinetis_probe (t=0x22000304) at target/kinetis.c:105
105                                     break;
(gdb) load
Loading section .text, size 0x10ef0 lma 0x0
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

At this point, gdb never returns.  I've tried waiting 10 minutes, doesn't matter. If I restart gdb and reconnect and compare sections, I can tell it was not successfully loaded:
```
(gdb) compare-sections
Section .text, range 0x0 -- 0x10ef0: MIS-MATCHED!
Section .ARM.exidx, range 0x10ef0 -- 0x10ef8: MIS-MATCHED!
Section .data, range 0x10ef8 -- 0x11920: MIS-MATCHED!
warning: One or more sections of the target image does not match
the loaded file

(gdb)
```
If I don't enable the target power, I get the same result. I don't think the target power should be necessary, but I'm not sure.

Schematics of the board may be found here:
http://www.ti.com/lit/ug/spmu296/spmu296.pdf

Schematics of the Tag Connect cable can be found here:
http://www.tag-connect.com/Materials/TC2050-IDC-NL%20Datasheet.pdf

BMPM2 Schematic:
https://github.com/blacksphere/blackmagic/wiki/files/bmpm_v2_0f_schematic.pdf

Electrically, I think this should work fine. Pins 3, 5, and 9 of the Tag Connect are grounded, as they are on the BMPM2 side of things. The JTAG TMS/TCK/TDO/TDI pins all look wired up correctly between the BMPM2, through the tag connect, and to the pins of the ICDI TM4C123 (see U2-A). The ICDI JTAG RST line is going to the RESET pin of the the ICDI TM4C123 (see U2-B).

Not sure what else to try here. Any suggestions?

gsmcmullin, 2017-01-31:
- If the scan and attach worked, things are connected and powered correctly.
- There may be issues with the TM4C driver, I haven't tested this myself.
- Once a timeout has occured and GDB start compaining with 'Ignoring packet error, continuing...' it's unlikely the session will recover.  You can set GDB's timeout to a large value with the command `set remotetimeout 100000`.  This may help if the only problem is that it's taking too long.
- You can issue `set debug remote 1` which will log the communication with BMP, so we can see which packet is causing the problem (if it's the flash erase or write).
- Is it possible that your device has some flash protection enabled that's preventing it from being erased or written?  There is no checking of error conditions in that driver.

esden, 2017-02-24:
@hazelnusse Were you able to gather more information on this? Is this issue resolved?

hazelnusse, 2017-03-22:
I haven't had time to pursue this; the last time I tried, I could not get it to work.

esden, 2017-03-22:
@hazelnusse thank you for the heads up. I have ordered the eval board you mentioned. I should get it here hopefully tomorrow so I can see myself if I can reproduce your issue.

gsmcmullin, 2017-04-12:
This should be fixed by #202.

closed 2017-04-12 by gsmcmullin