blackmagic-debug/blackmagic #190: Breaking into tight loops

UweBonnes, 2017-03-02:
Hello,

if  a step into a tight loop is made, like e.g.
while (systick_get_value() > (systick_value - 20000)); 
with systick not running, ^C does not break into the debugger. Is there any way to break into such loop?

DanielO, 2022-11-25:
I have reproduced this on the 1 bitsy board with the 'fancy blink' demo by replacing the 'nop' with a semicolon.

Load it then 'break main' and 'step' until the for loop then GDB will hang.

Killing gdb and reattaching shows it at the expected place, but another 'next' will hang it again.

When GDB is killed it prints the following:
```
[Havok 16:22] ~/projects/1bitsy-examples/examples/1bitsy/fancyblink >fg
[1]    continued  /Applications/ARM/bin/arm-none-eabi-gdb -ex  -ex 'mon tpwr enable' -ex  -ex
/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/src/gdb/gdb/thread.c:95: internal-error: struct thread_info *inferior_thread(): Assertion `current_thread_ != nullptr' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) Recursive internal problem.
```

I am using a:
  *  Black Magic Probe v1.8.2, Hardware Version 3
  * 1bitsy V0.1d
  * 1bitsy-examples 3d4c4f3e09094aab3dfb33554acb76f7233e9b45
  *  macOS 12.6
  * [ARM GNU toolchain 11.3-rel1](https://developer.arm.com/-/media/Files/downloads/gnu/11.3.rel1/binrel/arm-gnu-toolchain-11.3.rel1-darwin-x86_64-arm-none-eabi.pkg?rev=10957d7bd17e41a698273da79ee689e7&hash=CE34A8F6099D3C0985FAB0490E714C041EF3ACC5) (Has GDB 10.2.90.20210621-git in it)

DanielO, 2022-11-25:
Note that on the flip side changing my spin loop to add a NOP instruction works around the problem in my code.

dragonmux, 2022-11-25:
Just to check that we haven't inadvertently fixed this already, could you test this with the latest `main` firmware please?

dragonmux, 2022-11-25:
After talking about this on Discord, and testing `main` as requested, it has been found to be not-our-bug - this is GDB/GCC doing something funky. See https://gist.github.com/DanielO/22848bbaa64a76b5b37fda5603e54e01 for more information on what exactly is transpiring here.

The most relevant section of the Gist is:
```
Breakpoint 1, i2c_write7_v1 (i2c=1073763328, addr=80, data=0x2001ffa4 "", n=2, start=true) at hal.c:62
62	    if (start) {
(gdb) next
63		for (tout = I2C_TIMEOUT; tout > 0 && (I2C_SR2(i2c) & I2C_SR2_BUSY); tout--)
(gdb) set debug remote 1
(gdb) next
[remote] Sending packet: $Z1,8000546,2#7c
[remote] Received Ack
[remote] Packet received: OK
[remote] Sending packet: $m800054e,2#61
[remote] Received Ack
[remote] Packet received: 404b
[remote] Sending packet: $m800054e,2#61
[remote] Received Ack
[remote] Packet received: 404b
[remote] Sending packet: $m8000550,2#2d
[remote] Received Ack
[remote] Packet received: 7b61
[remote] Sending packet: $m800054e,4#63
[remote] Received Ack
[remote] Packet received: 404b7b61
[remote] Sending packet: $m800054e,2#61
[remote] Received Ack
[remote] Packet received: 404b
[remote] Sending packet: $m800054e,2#61
[remote] Received Ack
[remote] Packet received: 404b
[remote] Sending packet: $Z1,8000550,2#77
[remote] Received Ack
[remote] Packet received: OK
[remote] Sending packet: $c#63
[remote] Received Ack
[remote] wait: enter
[remote] wait: exit
[remote] wait: enter
  [remote] Packet received: T05
  [remote] select_thread_for_ambiguous_stop_reply: enter
    [remote] select_thread_for_ambiguous_stop_reply: process_wide_stop = 0
    [remote] select_thread_for_ambiguous_stop_reply: first resumed thread is Thread 1
    [remote] select_thread_for_ambiguous_stop_reply: is this guess ambiguous? = 0
  [remote] select_thread_for_ambiguous_stop_reply: exit
[remote] wait: exit
```

closed 2022-11-25 by dragonmux