blackmagic-debug/blackmagic #191: Sometimes Programming hangs on Nrf51

shirish47, 2017-03-10:
I am using black magic probe with my custom device based on MDBT40P nrf51 module. sometimes when I try to program my device programing fails , chip is detected but not programmed. I am using  http://www.rogerclark.net/arduino-on-the-nrf51822-bluetooth-low-energy-microcontroller/ method.

my arduino IDE shows this.
`Sketch uses 13,872 bytes (7%) of program storage space. Maximum is 196,608 bytes.
Forcing reset using 1200bps open/close on port /dev/cu.usbmodemDDE7BCE1
/Users/shirishbytelens/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gdb -quiet -cd /var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/build1d29b9407593efadd738026abe19248e.tmp -b 230400 -ex set debug remote 0 -ex set target-async off -ex set remotetimeout 60 -ex set mem inaccessible-by-default off -ex set confirm off -ex set height 0 -ex target extended-remote /dev/cu.usbmodemDDE7BCE1 -ex monitor swdp_scan -ex attach 1 -ex file Hardware_Test.ino.elf -ex load -ex tbreak main -ex run -ex echo \n\n\nUpload finished! -ex quit 
Target voltage: Not Implemented!
Available Targets:
No. Att Driver
1 Nordic nRF51 0x0084 
Remote debugging using /dev/cu.usbmodemDDE7BCE1
Attaching to Remote target
0xfffffffe in ?? ()
Reading symbols from /private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/build1d29b9407593efadd738026abe19248e.tmp/Hardware_Test.ino.elf...done.
Loading section .text, size 0x3630 lma 0x1c000
Loading section .ARM.exidx, size 0x8 lma 0x1f630
Loading section .data, size 0x78 lma 0x1f638
Start address 0x1ca24, load size 14000
Transfer rate: 22 KB/sec, 823 bytes/write.
Temporary breakpoint 1 at 0x1d90e: file C:\Users\rclark\Documents\Arduino\hardware\RBL\RBL_nRF51822\cores\RBL_nRF51822/main.cpp, line 8.
Starting program: /private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/build1d29b9407593efadd738026abe19248e.tmp/Hardware_Test.ino.elf 
Note: automatically using hardware breakpoints for read-only addresses.`

gsmcmullin, 2017-03-12:
That output suggests that programming was successful.  I'm not sure what you mean by stuck, but after the above output the target should be running.  Maybe @rogerclarkmelbourne can help.

rogerclarkmelbourne, 2017-03-13:
Looks like the ID code is being recognised as a nRF51.

@shirish47

How do you know its not being programmed?  You would need to read the flash to confirm whether its been programmed or not.

Otherwise the problem may be in your code.

Or perhaps you didnt flash the Softdevice (or flashed the wrong Softdevice)

shirish47, 2017-03-14:
No I do that same process on many devices many work fine. it happens on some devices only. I don't think anything is wrong with hardware. if you can tell me how to read flash with it. that would be great. I will check it with gdb. I flash these devices multiple times with s130 if it does not work and I try again. please let me know how to read flash. 👍

markrages, 2017-03-23:
I'm seeing similar problems.  The softdevice hex loads successfully, then while loading the application, something gets stuck.  I attached a debugger to the debugger, and got this trace:
```
0x08005288 in _gpio_get (gpios=16, gpioport=1073809408)
    at platforms/stm32/gpio.h:65
65		return (uint16_t)GPIO_IDR(gpioport) & gpios;
(gdb) bt
#0  0x08005288 in _gpio_get (gpios=16, gpioport=1073809408)
    at platforms/stm32/gpio.h:65
#1  swdptap_bit_in () at platforms/common/swdptap.c:55
#2  0x08005322 in swdptap_seq_in_parity (ret=ret@entry=0x20004548, ticks=3, 
    ticks@entry=32) at target/swdptap_generic.c:46
#3  0x0800281a in adiv5_swdp_low_access (dp=0x20000d80, RnW=<optimized out>, 
    addr=<optimized out>, value=0) at target/adiv5_swdp.c:157
#4  0x08002296 in adiv5_mem_read (ap=0x20000d60, dest=0x2000459c, 
    src=3758157296, len=1) at target/adiv5.c:514
#5  0x08005888 in target_mem_read32 (t=t@entry=0x40010814, 
    addr=addr@entry=3758157296) at target/target.c:394
#6  0x0800303e in cortexm_halt_poll (t=0x40010814, t@entry=0x20000e68, 
    watch=watch@entry=0x0) at target/cortexm.c:474
#7  0x080038e8 in cortexm_run_stub (t=t@entry=0x20000e68, 
    loadaddr=loadaddr@entry=536870912, r0=r0@entry=0, r1=r1@entry=0, 
    r2=r2@entry=0, r3=r3@entry=0) at target/cortexm.c:631
#8  0x08004d92 in nrf51_flash_write (f=<optimized out>, dest=<optimized out>, 
    src=<optimized out>, len=<optimized out>) at target/nrf51.c:243
#9  0x08005676 in target_flash_write (t=0x7c0, dest=239616, 
    src=0x20000192 <pbuf+18>, len=992) at target/target.c:216
#10 0x080040cc in handle_v_packet (
    packet=0x20000180 <pbuf> "vFlashWrite:3a800:\200\177", plen=1010)
    at gdb_main.c:425
---Type <return> to continue, or q <return> to quit---
#11 gdb_main_loop (tc=0xefeeffde, tc@entry=0x20000010 <gdb_controller>, 
    in_syscall=in_syscall@entry=false) at gdb_main.c:268
#12 0x080041a8 in gdb_main () at gdb_main.c:472
#13 0x08004a6e in main (argc=<optimized out>, argv=<optimized out>)
    at main.c:47
```
(This is a slightly customized probe, with firmware from a couple of months ago.  But the same thing happens with HEAD firmware on a real-deal BlackMagic mini.)

markrages, 2017-03-23:
When failing, it will fail repeatedly for a few hours, then work repeatedly for a few days.

markrages, 2017-03-23:
@shirish47 This gdb command will read an nRF51:

dump binary memory flashmem.bin 0 0x40000

This is for the 256k version (xxAC).  For the 128k version replace "0x40000" with "0x20000"

This is a binary dump. gdb can dump in other formats, see https://sourceware.org/gdb/onlinedocs/gdb/Dump_002fRestore-Files.html

shirish47, 2017-03-23:
Thanks for it . I will try it once I assemble a new board and if it does not work.

gsmcmullin, 2017-04-12:
Before #201, a communication error while programming the flash could result in an unhandled fault in the firmware resulting in an unresponsive BMP.  Since fixing this, programming works for me every time.

Please reopen the issue if the problem persists with the current firmware.

gsmcmullin, 2017-04-20:
@esden has seen this on v1.6-rc1

somecoder42, 2017-08-08:
I could not program my NRF52 at all with BMP. I tried gdb 7.12 and 8.0 and BMP 1.6.1 and master.
When I added the fix from #230 and it worked:
`t->target_options |= CORTEXM_TOPT_INHIBIT_SRST;` in `nrf51_probe()`
Please add this fix to BMP.

joeferner, 2017-09-03:
`t->target_options |= CORTEXM_TOPT_INHIBIT_SRST;` change worked for me as well

shirish47, 2017-09-26:
@joeferner  where to do that change ? 

Unfortunately this happens to some of our boards and not all. Some times simply heating the device with hot air helps to program but once it cools down this same issue persists.

```
Target voltage: Not Implemented!
Available Targets:
No. Att Driver
 1      Nordic nRF51 0x0084 
Remote debugging using /dev/cu.usbmodemDDE7BCE1
Attaching to Remote target
0x0001cc3c in ?? ()
Reading symbols from /private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_623610/Hardware_Test.ino.elf...done.
Loading section .text, size 0x3630 lma 0x1c000
Loading section .ARM.exidx, size 0x8 lma 0x1f630
Loading section .data, size 0x78 lma 0x1f638
Start address 0x1ca24, load size 14000
Transfer rate: 22 KB/sec, 823 bytes/write.
Temporary breakpoint 1 at 0x1d90e: file C:\Users\rclark\Documents\Arduino\hardware\RBL\RBL_nRF51822\cores\RBL_nRF51822/main.cpp, line 8.
Starting program: /private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_623610/Hardware_Test.ino.elf 
Note: automatically using hardware breakpoints for read-only addresses.

```

joeferner, 2017-09-26:
@shirish47  I added it right after https://github.com/blacksphere/blackmagic/blob/fd9efa27115cffe559cd21a9a40bde03a593f157/src/target/nrf51.c#L107

Then reflashed the blackmagic probe and haven't had a problem since then.

shirish47, 2017-09-26:
@joeferner I install blackmagic as guided by rogerclark I can't find this in my Mac book pro. you mean to compile blackmagic again and install it to my probe (stm32f103C)? 
if you already have a compile blackmagic please share it.

shirish47, 2017-09-27:
Well can anyone tell me why I am getting mergehex as conflict when I compile with arduino and try to merge s130.hex? I am using rogerClarks blog for this. One of my file does merge with the s130.hex, that file I compiled yesterday. but today I am not able to merge new hex file I compile with same s130.hex

siddhesh23sawant, 2018-10-10:
can anyone give me direct .bin file to program st-link , hope after reflashing i will not the get the stuck just like shirish47???
mail id siddhesh23sawant.ss@gmail.com

esden, 2021-12-15:
Based on the above comments and code history. This issue should be fixed with bea8436561124ab15158dfd13f5093ecd68c6363 I am closing it as this issue is getting pretty stale. If the programming is still problematic feel free to reopen this issue.

closed 2021-12-15 by esden