blackmagic-debug/blackmagic #192: DFU Upload from Bootloader area hangs

UweBonnes, 2017-03-13:
Running "dfu-util -s 0x08000000:0x2000 -U x" with present dfu-utils (March 2017) hangs. BMP bootloader stalls USB communication and that is not forseen in the DFUSE specs. While dfu-utils has a hotfix for that problem, I think it is better if the bootloader behaves right. Appended patch tried to behave in a compliant way.
The patch removes the range check in the "CMD_ERASE" case. I think this is also usefull to handle STM32F103C8 devices that nearly always have 128 kByte flash.

[Uploading 0001-Handle-SET_ADDRESS-according-to-DFUSE-specs.txt…]()

esden, 2017-04-12:
Where can we find the patch for this? Can you create a pull request for this? :)

UweBonnes, 2017-04-20:
See pull requesy #206

closed 2017-04-27 by gsmcmullin