blackmagic-debug/blackmagic #211: v1.6 release testing

gsmcmullin, 2017-04-18:
Tests:
- [ ] Scan and attach
- [ ] Load program to flash
- [ ] Verify flash programming
- [ ] Start (Temporary flash breakpoint and execution control)
- [ ] Memory inspection

Target hardware:
- [ ] 1Bitsy (STM32F4)
- [ ] Lisa/M v1 (STM32F1)
- [ ] EK-TM4C123GXL (TM4C)
- [ ] FRDM-KL03Z (KL03Z)

esden, 2017-04-19:
The list of targets that I have tested successfully:

Each test includes:
* Scan
* Attach
* Load
* Verify flash
* Run
* Interrupt
* Memory inspection

Hardware:
- [x] 1Bitsy (STM32F4)
- [x] Lisa/M V1 (STM32F1)
- [ ] EK-TM4C123GXL (TM4C)
- [x] FRDM-KL03Z (KL03Z)
- [ ] FRDM-KL22F (KL22F)
- [x] Adafruit Bluefruit LE (nrf51822)
- [ ] SparkFun nRF52832 Breakout (nRF52832)

esden, 2017-04-21:
Based on the information above I think we have good confidence that we can release the 1.6 stable version.

closed 2017-04-21 by gsmcmullin