blackmagic-debug/blackmagic #221: F7 and WFI

UweBonnes, 2017-05-01:
Hello Gareth,

I try to single step on an F767, mostly idling in WFI. There are several issues:
1., "mon s" does not find the device, until the device is in reset, either by the button or " mon connect_srst enable". Openocd has the same issue, but only reset button access is available at that stage.
2. Breakpoints are accepted after attach, but running results in error:
(gdb) mon connect_srst enable
"monitor" command not supported by this target.
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon s
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32F7xx
(gdb) att 1
Attaching to program: /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/bootloader.elf, Remote target
Stm32BootloaderCheck () at /home/bon/devel/ethernut_sf/nut/arch/cm3/dev/stm/stm32_bootloader.c:107
107     {
(gdb) b main
Breakpoint 1 at 0x80001e4: file /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/../bootloader.c, line 64.
(gdb) r
Starting program: /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/bootloader.elf 
Note: automatically using hardware breakpoints for read-only addresses.
Warning:
Cannot insert hardware breakpoint 1.
Could not insert hardware breakpoints:
You may have requested too many hardware breakpoints/watchpoints.

Command aborted.

After enabling Debug in Sleep (and Stop and standby) breakpoints work and work no longer with debug in sleep switched off again:
$2 = 2690580487
(gdb) p /x *(uint32_t*) 0xe0042004 = 0xa05f0007
$3 = 0xa05f0007
(gdb) b main
Breakpoint 1 at 0x80001e4: file /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/../bootloader.c, line 64.
(gdb) r
Starting program: /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/bootloader.elf 
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, main () at /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/../bootloader.c:64
64      {
(gdb) qu^CQuit
(gdb) p /x *(uint32_t*) 0xe0042004 = 0xa05f0000
$4 = 0xa05f0000
(gdb) r
Starting program: /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/bootloader.elf 
Warning:
Cannot insert hardware breakpoint 1.
Could not insert hardware breakpoints:
You may have requested too many hardware breakpoints/watchpoints.

Command aborted.

3.  Running without DBG_SLEEP results in unexpected SIGTRAP and SIGINT
 1      STM32F7xx
(gdb) att 1
Attaching to program: /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/bootloader.elf, Remote target
Stm32BootloaderCheck () at /home/bon/devel/ethernut_sf/nut/arch/cm3/dev/stm/stm32_bootloader.c:107
107     {
(gdb) r
Starting program: /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/bootloader.elf 

Program received signal SIGTRAP, Trace/breakpoint trap.
0xa05f0000 in ?? ()

4., After SIGTRAP, a power cycle of the F7 is needed.

What about unconditionally enable Debug in Sleep, as most OpenOCD start scripts do. Some discussion around that was in
https://sourceforge.net/p/blackmagicdebug/mailman/message/32867555/
Adding    target_mem_write32(t, DBGMCU_CR, DBG_STANDBY| DBG_STOP | DBG_SLEEP); after successfull probe solved the issues for me.

As DBGMCU_CR is not reset by reset, debug under reset is only needed for the first attach after a cold boot.

UweBonnes, 2017-05-01:
Here some debug output with a breakpoint set and DBGMCU_CR == 0:
(gdb) set debug remote 1
(gdb) r
Sending packet: $k#6b...Ack
Sending packet: $?#3f...Ack
Packet received: W00
Starting program: /home/bon/devel/ethernut_projects/bootloader/nucleo_f767zi/bootloader.elf 
Sending packet: $vRun;#e6...Ack
Packet received: T05
Sending packet: $qC#b4...Ack
Packet received: 
Sending packet: $qOffsets#4b...Ack
Packet received: 
Sending packet: $Hg0#df...Ack
Packet received: 
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/>    <reg name="r11" bitsize="32"/>    <reg name="r12" bitsize="32"/>    <reg name="sp" bitsize="32" type="data_ptr"/>    <reg name="lr" bitsize="32" type="code_ptr"/>    <reg name="pc" bitsize="32" type="code_ptr"/>    <reg name="xpsr" bitsize="32"/>    <reg name="msp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="psp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="special" bitsize="32" save-restore="no"/>  </feature></target>
Sending packet: $qXfer:features:read:target.xml:3cf,3fb#12...Ack
Packet received: l
Sending packet: $g#67...Ack
Packet received: 00005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa0
Sending packet: $qXfer:memory-map:read::0,3fb#e5...Ack
Packet received: m<memory-map><memory type="ram" start="0x20000000" length="0x80000"/><memory type="ram" start="0x00000000" length="0x4000"/><memory type="flash" start="0x00240000" length="0x1c0000"><property name="blocksize">0x40000</property></memory><memory type="flash" start="0x00220000" length="0x20000"><property name="blocksize">0x20000</property></memory><memory type="flash" start="0x00200000" length="0x20000"><property name="blocksize">0x8000</property></memory><memory type="flash" start="0x08040000" length="0x1c0000"><property name="blocksize">0x40000</property></memory><memory type="flash" start="0x08020000" length="0x20000"><property name="blocksize">0x20000</property></memory><memory type="flash" start="0x08000000" length="0x20000"><property name="blocksize">0x8000</property></memory></memory-map>
Sending packet: $qXfer:memory-map:read::322,3fb#4c...Ack
Packet received: l
Sending packet: $m80001e4,2#5d...Ack
Packet received: 0000
Sending packet: $G00005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa000005fa100005fa000005fa000005fa0#b8...Ack
Packet received: OK
Sending packet: $Z1,80001e4,2#a7...Ack
Packet received: E01
Warning:
Cannot insert hardware breakpoint 1.
Could not insert hardware breakpoints:
You may have requested too many hardware breakpoints/watchpoints.

Command aborted.

esden, 2021-12-15:
Is this still an issue? Does it need to be addressed? This issue is getting bit stale. :)

UweBonnes, 2022-01-25:
Fixed in current head

closed 2022-01-25 by UweBonnes