blackmagic-debug/blackmagic #224: flashing LPC812 gives MIS-MATCHED, vectors and such are correct

Squantor, 2017-05-04:
After flashing a small program into the LPC812 I get the mismatch error. Initially I suspected that the checksum vector was not filled in properly (the LPC series microcontrollers need a checksum in interrupt vector location 8). This is not the case, even more strange. The binary file from the compile and the dump with GDB are exactly the same! So I should not be getting this error.

I was not able to actually debug the black magic probe. I need to solder up some blue pill boards so I have a debugging the debugger hardware setup.

These are the commands I input:
```
target extended-remote /dev/ttyBmpGdb
monitor swdp_scan
attach 1
set mem inaccessible-by-default off
set {int}0x40048000 = 2
file periph_blinky.elf
load periph_blinky.elf
compare-sections
```

Result:
```
Section .text, range 0x0 -- 0x1080: MIS-MATCHED!
Section .ARM.exidx, range 0x1080 -- 0x1088: matched.
warning: One or more sections of the target image does not match the loaded file
```

Attached is the elf, map, lss files with the original (periph_blinky.bin) and read out (output.bin) binary. See and try for yourself, maybe one of you has a LPC812 target lying around and is capable of debugging this solution?

[LPC812 mismatched.zip](https://github.com/blacksphere/blackmagic/files/977532/LPC812.mismatched.zip)

gsmcmullin, 2017-05-04:
`load` resets the target, which will undo the effect of your `set {int}0x40048000 = 2`.  You'll need to do this again after a load.

Squantor, 2017-05-05:
@gsmcmullin That was it! Just tried it out, and works now. Its actually quite obvious now I think about it. My apologies for this case of "Problem exists between probe and lab chair" :)

closed 2017-05-05 by gsmcmullin