blackmagic-debug/blackmagic #225: gdb doesn't respond to interrupt

ladmanj, 2017-05-05:
When trying to use gdb with BMP gdbserver and eclipse cdt, gdb doesn't respond to interrupt command.
While looking at the interaction between eclipse and gdb i can see this
request:
99-interpreter-exec --thread-group i1 console kill
response:
&"Quit\n"
&"Quit\n"
&"Quit\n"
&"Cannot execute this command while the target is running.\n"
&"Use the \"interrupt\" command to stop the target\n"
&"and then try again.\n"
99^error,msg="Cannot execute this command while the target is running.\nUse the \"interrupt\" command to stop the target\nand then try again."
(gdb) 

This prevents also inserting a breakpoint while process is running, only the breakpoints inserted before 
30-exec-run --thread-group i1 (comming from .gdbinit file) are accepted.

gdb version:
GNU gdb (GNU Tools for ARM Embedded Processors) 7.10.1.20151217-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
bmp version:
Black Magic Probe (Firmware v1.6-rc0-257-gd6e2977) (Hardware Version 0)
Copyright (C) 2015  Black Sphere Technologies Ltd.

Thanks

ladmanj, 2017-05-07:
Hello
I have tried it once again with gdb from toolchain ppa:team-gcc-arm-embedded/ppa as recommended in youtube black magic probe tutorial.
--version
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 7.12.1.20170215-git

It seems that the running program can be successfully interrupted also from eclipse.
So maybe the problem was fixed somewhere between gdb 7.10 and 7.12, or is specific to ac6 system workbench patches.

closed 2017-05-07 by ladmanj

gsmcmullin, 2017-05-07:
Thanks, that's good to know.  There is, however a known problem that timeouts don't work correctly in gdb 7.12, so you may have issues particularly when flashing the target device.  This is fixed in GDB upstream, so hopefully when we have a gdb 8.0 everything will work.