blackmagic-debug/blackmagic #227: Extended reset gets called twice

falstaff84, 2017-05-08:
To support i.MX 7's Cortex-M4 I am using the extended_reset callback. Without the special handling, the CPU seems not to reset.

I am using a simple Hello World firmware on the target. When I do a simple `load`, break `main` and `r` cycles I see partial `Hello Worl` and then the debugger breaks. Setting a breakpoint in the extended_reset callback shows that a single `r` in gdb calls `extended_reset` twice (or actually reset for that matter):

```gdb
Breakpoint 3, imx7m4_extended_reset (t=0x20001ab0) at target/imx7.c:106
106     {
(gdb) bt
#0  imx7m4_extended_reset (t=0x20001ab0) at target/imx7.c:106
#1  0x080042da in cortexm_reset (t=0x20001ab0) at target/cortexm.c:452
#2  0x080054fc in gdb_main_loop (tc=tc@entry=0x20000000 <gdb_controller>, in_syscall=in_syscall@entry=0x0) at gdb_main.c:239
#3  0x080057f0 in gdb_main () at gdb_main.c:483
#4  0x08007350 in main (argc=<optimized out>, argv=<optimized out>) at main.c:47
(gdb) c
Continuing.
imx7m4_extended_reset

Breakpoint 3, imx7m4_extended_reset (t=0x20001ab0) at target/imx7.c:106
106     {
(gdb) bt
#0  imx7m4_extended_reset (t=0x20001ab0) at target/imx7.c:106
#1  0x080042da in cortexm_reset (t=0x20001ab0) at target/cortexm.c:452
#2  0x08005458 in handle_v_packet (packet=0x200009b0 <pbuf> "vRun;", plen=0x5) at gdb_main.c:410
#3  gdb_main_loop (tc=tc@entry=0x20000000 <gdb_controller>, in_syscall=in_syscall@entry=0x0) at gdb_main.c:279
#4  0x080057f0 in gdb_main () at gdb_main.c:483
#5  0x08007350 in main (argc=<optimized out>, argv=<optimized out>) at main.c:47
(gdb) 
```

Depending on the firmware/system this immediately aborted runs could be a problem... I removed the target_reset call at in the kill (`'k'`) case, which seems to avoid this problem but might cause other issues?

gsmcmullin, 2017-05-08:
Is this a problem?  Its a consequence of the fact that GDB first 'kills' the old inferior before starting a new one when you `run`.  This has always been the case.

I find it useful to be able to use `kill` to reset and detach.

From GDB protocol description for the `'k'` packet:
> Kill request.
> The exact effect of this packet is not specified.
> For a bare-metal target, it may power cycle or reset the target system. For that
> reason, the ‘k’ packet has no reply.

falstaff84, 2017-05-08:
Hm, I see the semantics of that packet is really not well defined.

It is mainly confusing, since when you set a breakpoint before `printf("Hello World");` then press `run`, you do not want to see any `Hello World` in your console (or a substring of it).... I was first thinking my breakpoint was somehow messed up, or source code/symbols miss match...

[OpenOCD seems to do nothing](https://github.com/ntfreak/openocd/blob/master/src/server/gdb_server.c#L2937) on kill and [this document](http://www.embecosm.com/appnotes/ean4/embecosm-howto-rsp-server-ean4-issue-2.html#id3033520) suggests so too.

esden, 2021-12-15:
This issue is becoming a bit stale. Do we have some actionable items on this? Should we keep this open? Should we close this?

agners, 2021-12-16:
This is related to #226. I stopped working with i.MX 7 devices a while ago, so from my side its fine to close this issue as well as the RFC PR.

esden, 2021-12-16:
Thank you @agners. In that case I am closing this. :)

closed 2021-12-16 by esden