blackmagic-debug/blackmagic #229: RTOS Threads support

tanshihaj, 2017-05-12:
I'm using ChibiOS in my projects. Before Black Magic Probe I've used OpenOCD + ST-Link and OpenOCD supports RTOS threads ([at least ChibiOS](http://wiki.chibios.org/dokuwiki/doku.php?id=chibios:community:setup:openocd_chibios)). It seems that Black Magic Probe doesn't support RTOS threads and you can see only one current thread (usually idle thread).

It will be great if Black Magic Probe begin to support at least ChibiOS threads.

I've found [mailing list](https://sourceforge.net/p/blackmagicdebug/mailman/message/32860779/) in which discussed this issue but it seems nothing changed.

UweBonnes, 2017-05-13:
This discussion has come up before. However the way it is done e.g. in OpenOCD, is regarded as layering Violation by Gareth. Other ideas are needed. As always: Actions speak louder than words...

gsmcmullin, 2017-05-15:
I don't believe that interpreting target data structures belongs in BMP firmware, but I'm leaving this issue open to track it.

IMO this issue should be addressed in GDB, probably by exposing an interface to allow a plugin to supply thread information.  See below for the discussion on the GDB mailing list.

A few references:
- "Python API for supplying thread information?" on the GDB mailing list:
https://sourceware.org/ml/gdb/2017-03/msg00033.html
- Fake ChibiOS 2 thread support from Python in GDB: https://github.com/gsmcmullin/gdb_chibios

UweBonnes, 2017-05-16:
Gareth, what is or was the state of "Fake ChibiOS 2 thread support from Python in GDB: https://github.com/gsmcmullin/gdb_chibios". Did it work at some point in time? How is it used?

gsmcmullin, 2017-05-16:
@UweBonnes It did work, and I have no reason to believe it doesn't work anymore, but I'm not currently using it.
- It is for ChibiOS 2 only, the data structures in ChibiOS 3 have changed.
- It was only ever used on STM32F4.  I think it should work with other ARMv7-M.
- It fakes thread support in GDB by overriding the `thread` and `info threads` commands from Python.
- Because of this it obviously won't work with frontends using the MI interface.

To use it you need to import the python modules into GDB.
- `python import chibios`
  This overrides `info threads` and `thread` commands in GDB to query threads and switch the active thread.
- `python import coredump`
  This overrides `gcore` to generate a core file with the current state of the target.

UweBonnes, 2018-03-27:
Gareth, does https://github.com/gsmcmullin/gdb_chibios need a python enabled gdb? Gdb as distributed by arm  https://developer.arm.com/open-source/gnu-toolchain/gnu-rm does not mention python with 
"arm-none-eabi-gdb --configuration"

gsmcmullin, 2018-03-28:
It does seem to support Python though:
```
gareth@gareth ~ $ arm-none-eabi-gdb
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) python print "hello"
hello
(gdb) 
```

UweBonnes, 2018-03-28:
My gdb behaves different, also the version string is the same:
`> arm-none-eabi-gdb *elf
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from caltime.elf...(no debugging symbols found)...done.
(gdb) python print "hello"
Python scripting is not supported in this copy of GDB.
(gdb) `

UweBonnes, 2018-05-15:
I looked deeper in gcc-arm-none-eabi-7-2017-q4-major/bin an found arm-none-eabi-gdb-py

sadziu82, 2020-08-14:
Hello,

I just made some small changes[1] to @gsmcmullin code[2] to make it working with ChibiOS 20.x. It looks like following:
```
(gdb) info threads
  Id   Target Id            Frame
  1    main                 0x80021A5 in chVTIsArmedI ()
* 2    idle                 0x8001C80 in _idle_thread ()
  3    rflinkPeripherial    0x80021A5 in chVTIsArmedI ()
```

1: https://github.com/gsmcmullin/gdb_chibios/pull/2
2: https://github.com/gsmcmullin/gdb_chibios