blackmagic-debug/blackmagic #230: Breakpoint not working on nRF52

gsmcmullin, 2017-05-15:
Target: nRF52832 DK (PCA10040)

BMP is able to scan, attach, and load to flash, but hardware breakpoint in flash is apparently never hit.

[dbg_out_backtrace.txt](https://github.com/blacksphere/blackmagic/files/1002243/dbg_out_backtrace.txt)

Confirmed:
- Addresses are in the flash memory region
- GDB is correctly sending the request for the hardware breakpoint
- Backtrace after interrupting with ctrl-c indicates a call from `main`, so target did start up correctly

Apart from the strangeness in that the breakpoint is never hit, the register read after reset (`vRun`) returns all zeros.  I suspected that the PC is being set to zero, and that the target would go directly to the fault handler when execution is resumed, but this appears to not be the case, given the backtrace.

MrZeroo00, 2017-05-15:
Binary is attached here
[nRF52_BMP.zip](https://github.com/blacksphere/blackmagic/files/1002275/nRF52_BMP.zip)
[Corresponding repo](https://github.com/MrZeroo00/nRF52_SDK13) commit 306b165d097be964894eae001f591fe08fe0b1aa

Thanks!

MrZeroo00, 2017-05-15:
As suggested @gsmcmullin on Gitter, adding `>target_options |= CORTEXM_TOPT_INHIBIT_SRST;` to nrf51_probe() did the trick.
It also seems to be flashing more consistently. I was getting very frequently a time out error when trying to load the binary

gsmcmullin, 2017-05-15:
From Gitter:

- Gareth McMullin @gsmcmullin 13:00
@MrZeroo00 It's a total guess, but my suspicion is that there is some timing issue with the target coming out of reset. I suggest try adding t->target_options |= CORTEXM_TOPT_INHIBIT_SRST; to nrf51_probe(). If that still doesn't work, try adding a delay here: https://github.com/blacksphere/blackmagic/blob/master/src/target/cortexm.c#L452
This may also be the cause of #191 that I can't reproduce.

- Badr Bouslikhin @MrZeroo00 13:28
@gsmcmullin the first suggestion worked!!
It also seems to be flashing more consistently. I was getting very frequently a time out when trying to load the binary

- Gareth McMullin @gsmcmullin 13:29
@MrZeroo00 Great!
That might be the #191 mentioned above.

- Badr Bouslikhin @MrZeroo00 13:31
Exactly, looks related to #191

joric, 2019-03-30:
I can't even flash nRF52840, it says:

```
Available Targets:                            
No. Att Driver                                
 1      ARM Cortex-M                          
 2      Nordic nRF52 Access Port              
 3      ARM Cortex-M                          
Attaching to Remote target failed             
Loading section .sec1, size 0xa000 lma 0x26000
Load failed                                   
The program is not being run.
```

I'm using
```
arm-none-eabi-gdb.exe --quiet --batch -ex "target extended-remote \\.\%port%" -ex "mon swdp_scan" ^
-ex "file %file:\=/%" -ex "att 1" %bmp_opt%

```
Maybe it's a known issue?

UweBonnes, 2019-03-30:
You should tell what verion of BMP you use!

joric, 2019-03-30:
```
arm-none-eabi-gdb.exe --batch -ex "target extended-remote \\.\%port%" -ex "mon version"

Black Magic Probe (Firmware v1.6.1-205-gc5c0783) (Hardware Version 0)        
Copyright (C) 2015  Black Sphere Technologies Ltd.                           
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
```
Hardware is 73-2G4M08S1C (nRF52840)-based from here https://github.com/joric/nrfmicro

UweBonnes, 2019-03-31:
Probably your Programm uses Long time Deep Sleep and it is hard to attach to such a device. If you only want to reprogramm, attach to the Access port, understand the available programs with "mon help" and erase the device. But as full erase will set the Input/Output voltage to 1.8 Volt, reattach after full erase will only work with an Adapter that allows  Level translation to 1.8 Volt. Attaching to the firts port with under reset may also work. If you want to debug a program in Deep sleep, Things are harder.

joric, 2019-03-31:
@UweBonnes nope the matter was write protection. Is there a command in BMP similar to nrfjprog --recover or it's required to have J-Link for that?

UweBonnes, 2019-03-31:
Attaching to the second port and mass erasing via the Access port is what to do in that situation.

joric, 2019-03-31:
Can't attach neither to 1 nor to 2 neither with reset nor without.

UweBonnes, 2019-03-31:
"mon erase_mass"  is not a command of port 2.

joric, 2019-03-31:
Worked with manual attach and "mon erase_mass":
```
(gdb) target extended-remote \\.\COM8                                                 
Remote debugging using \\.\COM8                                                       
(gdb) mon swdp_scan                                                                   
Target voltage: unknown                                                               
Available Targets:                                                                    
No. Att Driver                                                                        
 1      ARM Cortex-M                                                                  
 2      Nordic nRF52 Access Port                                                      
(gdb) attach 2                                                                        
Attaching to Remote target                                                            
warning: while parsing target description: no element found                           
warning: Could not load XML target description; ignoring                              
PC register is not available                                                          
(gdb) mon erase_mass                                                                  
```
Thanks! Sorry for the offtopic here.

UweBonnes, 2019-04-01:
I have to correct my last post.  "mon erase_mass" is the only available command on target 2. And after erasing with "mon erase_mass" on target 2, the next scan should recognise target 1 as  Nordic nRF52

valff, 2019-10-25:
Adding `t->target_options |= CORTEXM_TOPT_INHIBIT_SRST;` works for me as well. Can this fix be added to master? I maintain a utility for [Drone Embedded OS](https://drone-os.com/) that does basic  workflow operations using BMP as a debug probe. Currently it works with STM32. I want to add nRF52 support, but it doesn't work without this fix. Telling the users to use a forked firmware would be unfortunate.

UweBonnes, 2019-10-28:
Please tell exactly the sequence of command that lead you to the unexpected result and explain what you expect. Please tell: Does the firmware use long (deep) sleep? Does it remap JTAG pins? If you can have a look at #507 if it has influence on the behaviour.

The whole usage of SRST in  cortexm_reset() seems dubious to me. Either we have control over the device or we have not. Why do srst in that situation at all? System reset will change debug registers that we have set up before!

Can this be reproduced with some NRF52840 dongle? Than give me some binary so I can reproduce.

Cat-Ion, 2019-12-29:
Possibly related: I'm using an Adafruit feather with an NRF52840, and am also having issues with breakpoints. With a simple main.c:

    int main() {
        volatile unsigned long i = 0;
        for(i = 0; i < 10000000; i++) {
                i = i;
        }
    
        volatile int foo = 0;
        while(!foo) {
                ;
        }
        return foo;
    }

loading the program in gdb (8.3.0.20190709-git) with current BMP HEAD (`c3a3f77`) and doing `start` does not hit any breakpoints, and I have to hit ^C to halt the program.

Putting a breakpoint in flash before the delay loop escalates into the Hardfault_Handler (and I still have to ^C)

Putting a breakpoint in flash after the delay loop halts the program as expected.

`starti`, then `stepi` skips a lot of instructions:

    (gdb) starti
    The program being debugged has been started already.
    Start it from the beginning? (y or n) y
    Starting program: /home/apo/Projects/NRF/asm/armgcc/_build/nrf52840_xxaa.out 
    
    Program stopped.
    0x00000000 in __isr_vector ()
    (gdb) stepi
    0x00000304 in main () at main.c:4
    4               for(i = 0; i < 10000000; i++) {
    (gdb) p i
    $1 = 4351075
    (gdb) 

I've attached a file containing the elf and a listing: [nrf52840.tar.gz](https://github.com/blacksphere/blackmagic/files/4008133/nrf52840.tar.gz)

mubes, 2020-01-13:
I have tested with current head and this problem now appears fully resolved and I think the bug report can be closed. There is some slight complexity working with these chips in that if code protection is switched on then it is essential that the mass erase is done via the second DAP (the only command it will accept) otherwise flashing via the first port will not work - the consequence being that GDB is out of step with what is actually in the chip, leading to mass confusion and symptoms like those in Cat-Ions note above.

UweBonnes, 2020-04-14:
I could clearly reproduce the failing breakpoint behaviour on a NRF52840 dongle cleaned before programming a simple blinky with  nrf51_mdm_cmd_erase_mass clean device. #637 finaly sets CORTEXM_TOPT_INHIBIT_SRST for all NRF5 devices.

closed 2020-04-14 by UweBonnes