blackmagic-debug/blackmagic #232: [Blue Pill] Official build doesn't seem to work

dev-zzo, 2017-05-20:
When flashed to the "Blue Pill" board via SWD/STLink, the official binary build does not appear to work; the USB device is not recognized properly, when connecting a STLink to it the CPU shows up to be in lockup.

Is there anything I could try to fix the problem?

UweBonnes, 2017-05-20:
Probably your executable size is larger than 56 k and your bluepill only has a F103C8.  Most F103C8 are rely F103CB in disguise, but programming an F103C8 above 64 k needs some thinking and (yet) patched tools.

Did you try with a BMP version that did not hit the 56 k limit?

dev-zzo, 2017-05-20:
You are absolutely right, the build I have downloaded doesn't fit into the 56k limit; 0xA0 bytes too long. Though, I seem to have a board with 128k Flash (reported by ST tools at least), but the firmware fails in the same way on that one as well.

Are there any recommended build options I could use to reduce the image size and try it out?

UweBonnes, 2017-05-21:
Easiest: remove some probe for some devices you don' need. Otherwise, with the most recent bootloader and dfu-utils patched with  https://sourceforge.net/p/dfu-util/tickets/44/,things should also work. Or mayby with a debugger attached to the bluepill, look if you can uploade the executable direct.

S3ler, 2017-05-30:
Hi, i am new with the Black Magic Probe and SWD/JTAG/Stlink.
I uploaded the (self compiled) firmware based on this tutorial: 
- https://medium.com/@paramaggarwal/converting-an-stm32f103-board-to-a-black-magic-probe-c013cf2cc38c
- compiled commit aaa7b0e38ed450eaf56388af89daef912399a25a

I use Ubuntu 15.04
```
uname -a
Linux ThinkPad-X61-Tablet 4.8.0-52-generic #55~16.04.1-Ubuntu SMP Fri Apr 28 14:36:29 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

dmesg recognizes the device, but creates no Serial Port (neither ttyACM0 or ttyACM1):
```
~$ dmesg
(...)
[ 1247.348167] usb 5-1: USB disconnect, device number 6
[ 1252.996069] usb 5-1: new full-speed USB device number 7 using uhci_hcd
[ 1253.181096] usb 5-1: New USB device found, idVendor=1d50, idProduct=6017
[ 1253.181104] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1253.181109] usb 5-1: Product: Black Magic Probe (Upgrade)
[ 1253.181114] usb 5-1: Manufacturer: Black Sphere Technologies
[ 1253.181119] usb 5-1: SerialNumber: C2BFB807
```
dfu-util or dfu-tool cannot open the device:
```
~$ dfu-util --list
dfu-util 0.8
(...)
dfu-util: Cannot open DFU device 1d50:6017
```
```
~$ sudo dfu-tool list
(dfu-tool:3731): libdfu-WARNING **: Failed to set new device: replugged device is not DFU-capable
Gefunden 0483:2016 [v0.1]:
  Status:       cannot get device state: device sent more data than requested
```

Is my problem related to this issue?

Informatic, 2017-08-02:
Just a note:

I've successfuly flashed BlackMagic onto BluePill (STM32F103C8T6) by using random st-link and following openocd command:
`openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg -c 'flash bank stm32f1x.flash2 stm32f1x 0x08000000 131072 0 0 stm32f1x.cpu' -c 'init' -c 'reset halt' -c 'flash write_image erase blackmagic_dfu.bin 0x8000000' -c 'flash write_image erase blackmagic.bin 0x8002000' -c 'shutdown'`

This is a really horrible workaround for flash autoprobing (this essentially declares additional, bigger bank, which overlaps with the one that is created during flash probing), but worked fine for me.

PetteriAimonen, 2017-08-08:
One workaround would be to add a build configuration that doesn't include the DFU bootloader - saves 8 kB and makes the rest of it fit. Because the blue pill has SWD and serial bootloader connections, the DFU bootloader is not very important.

closed 2017-10-08 by gsmcmullin