blackmagic-debug/blackmagic #234: Load fails on STM32F746

elagil, 2017-05-25:
Hello!

I am having problems loading an .elf to the STM32F746. It hangs after beginning to load the sections. I use the following commands:

`$ arm-none-eabi-gdb
target extended-remote <COM-port>
monitor swdp_scan
attach 1
file flash.elf
load`

Then, it starts loading the .elf:

`Loading section .isr_vector, size 0x10c lma 0x8000000`

but does not get past it. I do not get any error message, it just hangs there forever. Also, if I stop the command (SIGINT), the blackmagic probe does not react to further commands. I can flash an STM32F1 without problems an debug it.

I can mass erase the processor, however, with:

`mon erase_mass`

Thanks in advance!

elagil, 2017-05-29:
So, when I disconnect the NRST line, it works. Why is it so? It worked with the NRST connected when using an STM32F1.

UweBonnes, 2017-05-29:
What BMP platform do you use? Why is NRST asserted at all? NRST assertion need some commands from your side, as far as I know..

elagil, 2017-05-30:
I actually implemented the BMP V2.1 on my own PCB. The RST pin is just directly connected to PA2 of the STM32F103 on the BMP, just like it is in the reference design.

NRST is held low by the BMP all the time while debugging, as far as I can measure.

I use no special commands for assertion of NRST. You see the commands I used in the opening statement.

elagil, 2017-05-30:
I noticed that I missed the inverter in the original BMP design. Sorry for causing a stir, that was stupid.

gsmcmullin, 2017-05-30:
NRST is pulsed by `cortexm_reset` for targets where it is implemented correctly and does not pull NTRST with it.  It sounds like this is just an error in your hardware design, so I'm closing the issue.  Please reopen if that is not the case.

closed 2017-05-30 by gsmcmullin