blackmagic-debug/blackmagic #236: STM32 Timer interrupts disabled during debug

elagil, 2017-06-02:
Hello!

I have a simple timer interrupt running on a STM32F7. It triggers just fine, unless I am in a gdb debug session. Is this intended, to make single-stepping usable?

I connect to the STM like this:

``
mon swdp_scan
attach 1
set mem inaccessible-by-default off
file ./build/exec.elf
load
compare-sections
``

and it loads everything without problems.

``
load
Loading section .isr_vector, size 0x1c8 lma 0x8000000
Loading section .text, size 0x3a30 lma 0x80001c8
Loading section .rodata, size 0x44 lma 0x8003bf8
Loading section .init_array, size 0x4 lma 0x8003c3c
Loading section .fini_array, size 0x4 lma 0x8003c40
Loading section .data, size 0x18 lma 0x8003c44
Start address 0x8003a1c, load size 15452
Transfer rate: 15 KB/sec, 735 bytes/write.
compare-sections
Section .isr_vector, range 0x8000000 -- 0x80001c8: matched.
Section .text, range 0x80001c8 -- 0x8003bf8: matched.
Section .rodata, range 0x8003bf8 -- 0x8003c3c: matched.
Section .init_array, range 0x8003c3c -- 0x8003c40: matched.
Section .fini_array, range 0x8003c40 -- 0x8003c44: matched.
Section .data, range 0x8003c44 -- 0x8003c5c: matched.
Note: automatically using hardware breakpoints for read-only addresses.
Breakpoint 
1, main () at Src/main.c:114
continue
``

When I `load` and then `continue`, the interrupts do not trigger. They only come alive when the STM is power-cycled.

Thank you in advance for your comments!

UweBonnes, 2017-06-03:
read about dbgmcu functionality in the reference Manual and check the Settings there.

elagil, 2017-06-06:
So, I am using the standard settings, which enable all clocks by default. It is not a problem with BMP or the settings on the MCU, but rather the debugger that I use with Visual Studio code. It sends incorrect commands.

closed 2017-06-20 by gsmcmullin