blackmagic-debug/blackmagic #237: STM32F7 erase option bytes - load fails

elagil, 2017-06-02:
Hello!

I erased the option bytes on a STM32F7 with the following result:

```
mon opt erase
0x1FFF0000: 0x0000
0x1FFF0008: 0x0000
0x1FFF0010: 0x0000
0x1FFF0018: 0x0000
```

That is not in tune with the default settings that the reference manual states. How can I restore them? When using the `mon opt write <value>` command, the option bytes are not changed, regardless of what I enter as a value. Loading now fails like this:

```
load
Loading section .isr_vector, size 0x1c8 lma 0x8000000
Error writing data to flash
Error writing data to flash
```

UweBonnes, 2017-06-03:
please look at my recent pull request!

elagil, 2017-06-06:
I tested your pull request but it does not seem to work for my case.

```
(gdb) mon opt write C0FFAAFF 00400080
OPTCR: 0x00000000 OPTCR1: 0x00000000
```

The option bytes remain unchanged

Erasing also does not do the trick:

```
(gdb) mon opt erase
OPTCR: 0x00000000 OPTCR1: 0x00000000
```

UweBonnes, 2017-06-06:
What device do you use? Is perhaps some protection set? Can you reset the device with some other debugger? And then test from a clean start?

elagil, 2017-06-06:
I use an STM32F746. Protection is not set. I could reset it with the ST Link and ST Link Utility. After this, I can still not change any option bytes.

For example:
```
(gdb) mon option write 0xC0FFAAF3 0x00400080
OPTCR: 0xC0FFAAFF OPTCR1: 0x00400080
```

UweBonnes, 2017-06-06:
Please have a look at the STM32f74x reference manual, section 3.4.  In the sequence above, you only change th Brown-Out level (bit 2/3 in OPTCR) from the default level. Did the brown out level change after the reset after programming? And did "mon option erase" (and not "mon erase") return to the default brown out level?

elagil, 2017-06-06:
I know about the change of the brown out level. This was just for testing.

So, I can change the brown out level:
```
(gdb) mon option write 0xC0FFAAF3 0x00400080
OPTCR: 0xC0FFAAF3 OPTCR1: 0x00400080
```

Now, I try this:
```
(gdb) mon option erase
```

but after power cycle, the option bytes do not return to default.

If I use this:
```
(gdb) mon option write 0xC0FFAAFF 0x00400080
```
they do.

Also, when I type `mon option erase`, the debug probe does not blink the LED.

UweBonnes, 2017-06-06:
I misshandled the OPTSTART and OPTLOCK bits. Please try we a refreshed pull from my repo.

elagil, 2017-06-09:
I can confirm that it now restores the option bytes to:

```
OPTCR: 0xC0FFAAFC OPTCR1: 0x00400080
```

gsmcmullin, 2017-06-20:
This should be fixed by #235, now merged.

closed 2017-06-20 by gsmcmullin