blackmagic-debug/blackmagic #238: Build fails: this statement may fall through

a-v-s, 2017-06-06:
When trying to build the current code (aaa7b0e38ed450eaf56388af89daef912399a25a) the build fails with "error: this statement may fall through." I am using gcc version 7.1.0

```
...
  CC      gdb_main.c
gdb_main.c: In function 'gdb_main_loop':
gdb_main.c:157:16: error: this statement may fall through [-Werror=implicit-fallthrough=]
    single_step = true;
                ^
gdb_main.c:159:3: note: here
   case 'c': /* 'c [addr]': Continue [at addr] */
   ^~~~
gdb_main.c:167:16: error: this statement may fall through [-Werror=implicit-fallthrough=]
    single_step = false;
                ^
gdb_main.c:169:3: note: here
   case '?': { /* '?': Request reason for target halt */
   ^~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:71: gdb_main.o] Fout 1
make: *** [Makefile:9: all] Fout 2

```

UweBonnes, 2017-06-06:
Where does the "-Werror=implicit-fallthrough=" come from?

a-v-s, 2017-06-06:
-Wimplicit-fallthrough is enabled by -Wextra in the makefile. This is new in gcc7.

UweBonnes, 2017-06-07:
https://dzone.com/articles/implicit-fallthrough-in-gcc-7 explains the issue. Either BMP code disables the warning, or a lot of /*Fallthrough*/ comments have to be added. Gareth, what is your opinion?

esden, 2017-09-08:
I vote for using the explicit fall through comments as suggested by @tsaitgaist #273 PR. It is better to catch an implicit fall through in cases that it is not intended and have the ones that we want marked explicitly as such.

cpavlina, 2017-10-08:
Here's an updated PR with no merge conflicts, can we please merge this before more come up? I can't build without it.

gsmcmullin, 2017-10-08:
Fixed in 9a5b31c37bd5b55e8dc79d894f19aedd503c373c on master

closed 2017-10-08 by gsmcmullin