blackmagic-debug/blackmagic #242: SWD: Second load fails

elagil, 2017-06-09:
Hello!

I am using a self-built BMP 2.0 with a STM32F746 over SWD. I can load a flash file just fine, but if I want to load it a second time, it fails. I have to manually press and release the RST pin on my STM32. It seems like the NRST line is never pulled low on the BMP, so the processor is not reset.

```
$ arm-none-eabi-gdb
(gdb) target extended-remote \\.\COM10
(gdb) mon swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F7xx
(gdb) attach 1
(gdb) file test.elf
(gdb) load
Loading section .isr_vector, size 0x1c8 lma 0x8000000
Loading section .text, size 0x3f78 lma 0x80001c8
Loading section .rodata, size 0x64 lma 0x8004140
Loading section .init_array, size 0x4 lma 0x80041a4
Loading section .fini_array, size 0x4 lma 0x80041a8
Loading section .data, size 0x1c lma 0x80041ac
Start address 0x8003f68, load size 16840
Transfer rate: 14 KB/sec, 765 bytes/write.
(gdb) load
Loading section .isr_vector, size 0x1c8 lma 0x8000000
Error writing data to flash
(gdb) load
Error erasing flash with vFlashErase packet
```

Can I somehow force the BMP to pull NRST low on load? `mon connect_srst enable` does not do it.

UweBonnes, 2017-06-09:
I can not reproduce with the F7_discovery:
> arm-none-eabi-gdb *elf
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 7.12.1.20170215-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from uart_app.elf...(no debugging symbols found)...done.
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon s
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32F7xx
(gdb) att 1
Attaching to program: /devel/uart_dma/f7_discovery/uart_app.elf, Remote target
0xfffffffe in ?? ()
(gdb) load
Loading section .text, size 0x50fc lma 0x8000000
Loading section .data, size 0x2b8 lma 0x80050fc
Loading section .ram_func, size 0x18 lma 0x80053b4
Loading section .ARM.exidx, size 0x8 lma 0x80053cc
Start address 0x800007c, load size 21460
Transfer rate: 17 KB/sec, 858 bytes/write.
(gdb) load
Loading section .text, size 0x50fc lma 0x8000000
Loading section .data, size 0x2b8 lma 0x80050fc
Loading section .ram_func, size 0x18 lma 0x80053b4
Loading section .ARM.exidx, size 0x8 lma 0x80053cc
Start address 0x800007c, load size 21460
Transfer rate: 17 KB/sec, 858 bytes/write.
(gdb) load
Loading section .text, size 0x50fc lma 0x8000000
Loading section .data, size 0x2b8 lma 0x80050fc
Loading section .ram_func, size 0x18 lma 0x80053b4
Loading section .ARM.exidx, size 0x8 lma 0x80053cc
Start address 0x800007c, load size 21460
Transfer rate: 17 KB/sec, 858 bytes/write.
(gdb)

elagil, 2017-06-09:
That is interesting. Can you also `load` without problems after `continue`ing? I am not sure where to trace that issue. Could it actually be the binary file I use? It does not use the SWD pins for anything else, as far as I know.

UweBonnes, 2017-06-09:
If you run something that mostly idles in WFI/WFE, either set the DBGMCU_CR register to  DBG_STANDBY| DBG_STOP | DBG_SLEEP or you need  to set "mon connect_srst enable"  with that functionality working...

gsmcmullin, 2017-06-20:
The target is reset just before the flash erase: https://github.com/blacksphere/blackmagic/blob/master/src/gdb_main.c#L424

That reset does pulse the SRST signal (except for on targets where this is known to cause problems) https://github.com/blacksphere/blackmagic/blob/master/src/target/cortexm.c#L432

elagil, 2017-06-21:
Ok, I can actually see the pulse on the scope. 

But, I only see it every second time I use flash, which is very odd. When the reset does NOT happen, flashing actually works.

I have a 100n CAP on the NRST pin, so it takes 4 ms until 50% VDD is reached again and around 12 ms until it is 90% VDD. How much does it wait for flashing after reset? Maybe the target is still in the reset state when the flashing starts.

gsmcmullin, 2017-06-21:
It doesn't wait a prescribed amount of time, but polls for `S_RESET_ST` in `DHCSR`:
https://github.com/blacksphere/blackmagic/blob/master/src/target/cortexm.c#L452

This was intended to wait until the core is released from reset.  This might be incorrect, and this flag may be set while still in reset.  We should possibly add a poll on the SRST signal too.

elagil, 2017-06-22:
Without the 100n capacitor on reset, it works all the time. Adding a poll to the SRST pin makes sense, in my opinion. 

The reference manual for the STM32F7 states that 100n should be added to the NRST pin, no external pullup required. Therefore, I assume that my configuration is quite common.

UweBonnes, 2017-06-22:
Elagil,

please set DBGMCU_CR to 7 and try again. I am quite sure this is a WFI/WFE problem.

elagil, 2017-06-22:
So, I put in the 100nF again and added

`DBGMCU->CR |= DBGMCU_CR_DBG_SLEEP | DBGMCU_CR_DBG_STOP | DBGMCU_CR_DBG_STANDBY;`

to the `main()`. This, again, fails at every second load attempt (when SRST is pulled low). I actually need to remove the capacitor.

UweBonnes, 2017-06-22:
I just work with a NUCLEO_F767 that has the same 100 nF capacitor on the reset line. I flash and reflash again and again. But I have set
- mon connect_srst enable
and after attach:
- p /x *(uint32_t*) 0xE0042004 = 7

elagil, 2017-06-22:
I did exactly the same now. However, I still get:

`Error writing data to flash`

Do you have the reset line connected? If completely removed, it should also work.

esden, 2021-12-15:
What is the status on this? @elagil were you able to sort your issue out? As far as I understand this is mostly related to your personal hardware setup and not a BMP issue directly? Unless the BMP reset pulse is too short? I am asking because this issue is becoming bit stale. :)

elagil, 2021-12-16:
@esden I did not investigate the issue further. I agree that it was probably related to my particular hardware setup, and for that moment, I was satisfied with the solution of removing the capacitor.

esden, 2021-12-16:
Ok, sounds good. Thank you very much for the update. Maybe at some point we should address this specifically as an `enhancement` to allow for very long reset timings. But for now I will close this issue, as your problem is resolved. :)

closed 2021-12-16 by esden

UweBonnes, 2021-12-16:
The STM32F7 under WFI was causing a lot of grief. It is hard to "warm attach" when WFI was used. With  repeated writes via TRNCNT to DHCSR most probes now attach consistant in short time. Git head has the needed changes, https://github.com/UweBonnes/blackmagic/tree/fixes clean up and layers better and is to be applied soon.