blackmagic-debug/blackmagic #243: How to load firmware into target quickly?

trlsmax, 2017-06-15:
Hi devs,
Thanks to make all the good work on the project.
After compile and load it into my stlink clone, I can debug with gdb and use the vcp.
But how to load firmware into target quickly ?
I know how to do it as the doc said:
1.connect to target
2.open gdb
3.connect to vcp
4.attach to target
5.open firmware in gdb
6.load firmware into target

but how to make it quick? something like st-flash command ?

esden, 2017-06-15:
You can automate the process. Take a look at how it is done in the https://github.com/1Bitsy/1bitsy-locm3-template .

You can use a script https://github.com/1Bitsy/1bitsy-locm3-template/blob/master/scripts/black_magic_probe_flash.scr

Here is an abbreviated version of a flash GDB script:
```
#monitor jtag_scan
monitor swdp_scan
attach 1
load
compare-sections
kill
```
and invoke gdb to run the flash process in one shell command:
```
arm-none-eabi-gdb -nx --batch -ex 'target extended-remote /dev/ttyACM0' -x black_magic_probe_flash.scr yourbinary.elf
```
Here is an example of how to autodetect the BMP port and execute flash as part of a Makefile: https://github.com/1Bitsy/1bitsy-locm3-template/blob/master/libopencm3.rules.mk#L220

I hope this helps.

trlsmax, 2017-06-15:
Thank you very much. It works.

closed 2017-06-15 by trlsmax

esden, 2017-08-24:
Just for reference, as some people might end up here searching for a solution. I have put together the above information plus a bunch more in the Black Magic Probe wiki: https://github.com/blacksphere/blackmagic/wiki/GDB-Automation

UweBonnes, 2017-08-24:
Jtag/SWD scan for targets is missing above!

esden, 2017-08-25:
@UweBonnes I have edited the initial comment to add the JTAG/SWD scan. (hopefully that will prevent confusion if someone is reading it some time in the future 😁 )