blackmagic-debug/blackmagic #249: When target stuck in WFI, synchronisation is lost between GDB and BMP

schodet, 2017-06-19:
Configuration : BMP 2.1c connected to USB, 1bitsy 1.0d connected using the debug connector, no USB on 1bitsy.

For some reason, BMP does not manage to interrupt the target, GDB tries to communicate with BMP without success.

Program is derived from 1bitsy-locm3-template, with the following diff:

```diff
diff --git a/src/main.c b/src/main.c
index e5bc2b5..302311c 100644
--- a/src/main.c
+++ b/src/main.c
@@ -42,6 +42,7 @@ static void gpio_setup(void)
 int main(void)
 {
        int i;
+       __asm__ __volatile__ ("wfi");

        clock_setup();
        gpio_setup();
```

Here is GDB session log:

```
GNU gdb (7.7.1+dfsg-1+6) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from src/1bitsy-example.elf...done.
(gdb) target extended-remote /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7BAE5BAD-if00
Remote debugging using /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7BAE5BAD-if00
(gdb) mon connect_srst enable
(gdb) mon tpwr enable
(gdb) mon swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F4xx
(gdb) attach 1
Attaching to program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf, Remote target
0x08000544 in reset_handler () at ../../cm3/vector.c:71
71                      *dest = *src;
(gdb) load
Loading section .text, size 0x65c lma 0x8000000
Loading section .data, size 0xc lma 0x800065c
Start address 0x8000534, load size 1640
Transfer rate: 3 KB/sec, 546 bytes/write.
(gdb) set debug remote
Argument required (integer to set it to.).
(gdb) set debug remote 1
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Sending packet: $vKill;1#6e...Ack
Packet received:
Packet vKill (kill) is NOT supported
Sending packet: $k#6b...Ack
Sending packet: $?#3f...Ack
Packet received: W00
Starting program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf
Sending packet: $vRun;#e6...Ack
Timed out.
Timed out.
Timed out.
Ignoring packet error, continuing...
Packet vRun (run) is supported
Sending packet: $qC#b4...Sending packet: $qC#b4...Sending packet: $qC#b4...Sending packet: $qC#b4...Packet received: E01
Sending packet: $qAttached#8f...Packet instead of Ack, ignoring it
Packet instead of Ack, ignoring it
Packet instead of Ack, ignoring it
Ack
Packet received:
Packet qAttached (query-attached) is NOT supported
Sending packet: $qOffsets#4b...Packet instead of Ack, ignoring it
Packet instead of Ack, ignoring it
Packet instead of Ack, ignoring it
Ack
Packet received:
Sending packet: $Hg0#df...Packet instead of Ack, ignoring it
Packet instead of Ack, ignoring it
Packet instead of Ack, ignoring it
Ack
Packet received:
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received:
Protocol error: qXfer:features:read (target-features) conflicting enabled responses.
Sending packet: $g#67...Ack
Packet received:
Bad register packet; fetching a new packet
Packet received: E01
Remote 'g' packet reply is of odd length: E01
Sending packet: $g#67...Ack
Packet received: EFF
Remote failure reply: EFF
(gdb) quit
A debugging session is active.

        Inferior 1 [Remote target] will be killed.

Quit anyway? (y or n) y
Sending packet: $k#6b...Ack
Sending packet: $?#3f...Packet instead of Ack, ignoring it
Ack
Packet received: W00
```

JojoS62, 2018-02-08:
Is there already a patch for V1.6 for the wfi problem (on STM32F4) ?

UweBonnes, 2018-06-22:
Can a reset with current git head be done?

schodet, 2018-06-25:
This bug is actually not about not being able to connect but about the desynchronization with gdb when there is a connection problem.

UweBonnes, 2019-08-20:
To counteract this problem, target specific things need to be done. For STM32, some bits on DBGMCU must be set. This is partial already done in Git Head.(Aug 20, 2019). My "cortexm-branch" works this out a little further. Please test, if your problem is with STM32.

Otherwise user contribution is needed.

l29ah, 2020-11-23:
@UweBonnes what branch? https://github.com/UweBonnes/blackmagic/branches doesn't have it.

UweBonnes, 2020-11-23:
It is now in master.

esden, 2021-12-15:
@schodet did you have a chance to test this? Can we close this issue?

schodet, 2021-12-17:
I can test this after my holydays.

esden, 2021-12-17:
@schodet sounds good. May your holidays be enjoyable and relaxing! :D

schodet, 2022-01-04:
I do not manage to interrupt it, but the synchronization is no longer lost.

```
nicolas@boktor:~/work/1bitsy/1bitsy-locm3-template% gdb-multiarch src/1bitsy-example.elf
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from src/1bitsy-example.elf...
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon version
Black Magic Probe v1.7.1-332-g8def28d, Hardware Version 3
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) mon connect_srst enable
Assert SRST during connect: enabled
(gdb) mon tpwr enable
Enabling target power
(gdb) mon swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F40x M4
(gdb) attach 1
Attaching to program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf, Remote target
0x00000000 in ?? ()
(gdb) load
Loading section .text, size 0x6cc lma 0x8000000
Loading section .data, size 0xc lma 0x80006cc
Start address 0x08000558, load size 1752
Transfer rate: 6 KB/sec, 584 bytes/write.
(gdb) set debug remote 1
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Sending packet: $vKill;1#6e...Ack
Packet received:
Packet vKill (kill) is NOT supported
Sending packet: $k#6b...Ack
Starting program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf
Sending packet: $vRun;#e6...Ack
Packet received: T05
Packet vRun (run) is supported
Sending packet: $qC#b4...Ack
Packet received:
Sending packet: $qAttached#8f...Ack
Packet received:
Packet qAttached (query-attached) is NOT supported
Sending packet: $qOffsets#4b...Ack
Packet received:
Sending packet: $Hg0#df...Ack
Packet received:
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/[508 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11...putpkt: Junk:
Ack
Packet received: mme="faultmask" bitsize="8" save-restore="no"/>    <reg name="control" bitsize="8" save-restore="no"/>  </feature>  <feature name="org.gnu.gdb.arm.vfp">    <reg name="fpscr" bitsize="32"/>    <regname="d0" bitsize="64" type="float"/>    <reg name="d1" bitsize="64" type="float"/>    <reg name="d2" bitsize="64" type="float"/>    <reg name="d3" bitsize="64" type="float"/>    <reg name="d4" bitsize="64" type="float"/>    <reg name="d5" bitsize="64" type="float"/>    <reg name="d6" bitsize="64" type="float"/>  [439 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:7b1,3fb#e0...Ack
Packet received: l
Sending packet: $g#67...Ack
Packet received: 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Sending packet: $P10=00000001#6f...Ack
Packet received: OK
Sending packet: $vCont?#49...Ack
Packet received:
Packet vCont (verbose-resume) is NOT supported
Sending packet: $Hc0#db...Ack
Packet received:
Sending packet: $c#63...Ack
^Cremote_pass_ctrlc called
remote_interrupt called
^Cremote_pass_ctrlc called
The target is not responding to interrupt requests.
Stop debugging it? (y or n) y
Disconnected from target.
```

schodet, 2022-01-04:
Debug bits in DBGMCU_CR have been removed in 5e8c8cae10423c9f42bf51fb83c3e982caf584da, then added back only for F7 in 5918608156098db4b4f42f386b59e75bef79b79a, then this has moved to `adiv5.c` in 9bb2807706f44ecbafc501f1b5b0f05670853e58.

Do you know the reason?

UweBonnes, 2022-01-07:
Sorry for the delay, did not notice until now.

The F7 is very unwilling to halt when mostly in WFI. For a hot plug, you need to halt the CPU. Fast repeated writing on DHCSR with the halt request can eventually catch the F7 in a wake moment. However slow debuggers, as CMSIS_DAP-HID, write at low rate and so oftend do not catch the CPU  in a wake moment. Then the idea came up that using TRNCNT  will write the request to DHCSR in a much higher rate than any debugger can achieve. But thansome debuggers (CMSIS) did not like high TRNCNT, so more fiddling was needed. Now we are in a state where even a slow debugger will halt the F7 with great probability. As long as a wakeup happens in the waiting time.

In all, this is a long story of try and error.

Patch F5e8c8ca only fixed the low power effect, at the expense of the hot plug, so more work was needed,

Finally cf41a60b from the fixes branch will move the target specific F7 handling from adiv5.c to stm32f4.c to improve layering. DBGMCU is reset an exit.  

I fear however the LPC55 will again need target specific code in adiv5.c...

schodet, 2022-01-10:
Tested with `v1.7.1-340-g17c5ae2`, I am still not able to break a program stuck in WFI.

I tried to read the register:

```
(gdb) mon version
Black Magic Probe v1.7.1-340-g17c5ae2, Hardware Version 3
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon tpwr enable
Enabling target power
(gdb) mon sw
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F40x M4
(gdb) attach  1
Attaching to program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf, Remote target
0x00000000 in ?? ()
(gdb) set mem inaccessible-by-default off 
(gdb) x/wx 0xE0042004
0xe0042004:     Cannot access memory at address 0xe0042004
(gdb) load
Loading section .text, size 0x6cc lma 0x8000000
Loading section .data, size 0xc lma 0x80006cc
Start address 0x08000558, load size 1752
Transfer rate: 6 KB/sec, 584 bytes/write.
(gdb) x/wx 0xE0042004
0xe0042004:     0x00000000
(gdb)
```

I tried to write it myself, but still stuck:

```
(gdb) set *(volatile unsigned int *)0xE0042004 = 7
(gdb) x/wx 0xE0042004
0xe0042004:     0x00000007
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf 
^C^CThe target is not responding to interrupt requests.
```

UweBonnes, 2022-01-10:
For me a run on a F429 look different:
```(gdb) tar ext /dev/ttyACM1
Remote debugging using /dev/ttyACM1
(gdb) mon
Black Magic Probe(STLINK-V2) v1.7.1-340-g17c5ae28, Hardware Version 1
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) mon s
Target voltage: 2.91V
Available Targets:
No. Att Driver
 1      STM32F40x M4
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x08001f84 in ?? ()
(gdb) x /x 0xE0042004
0xe0042004:     0x00000007
```
DBGMCU_CR is set by default.  Be sure to unpower the CPU after some mishap has happened before. The debug unit may be screwed up from that mishap. Additional, please run hosted to see if there are any unexpected messages.

Setting DBGMCU on probe is now PR#973.

Attach breaking at address 0 in your example above seems dubious. Perhaps the program is not valid and and ther is immediate a double fault or similar. Try also connect under reset or after a mass erase.

UweBonnes, 2022-01-10:
Okay: Steps to reproduce:
- programm nonsense to the flash so it crashes after startup.
- Use a firmware that does not set DBGMCU_CR_SLEEP etc in probe
- Attach to the device after powerup, so DBGMCU_CR is noot set from tests before
Connecting under reset will allow to read DBGMCU_CR after attach.

Please test #973

schodet, 2022-01-12:
Using #973:
- load a program without WFI,
- I can debug, break and read the DBGMCU_CR register,
- add a WFI right at the begin of main, rebuild,
- load,
- run,
- can not break (`The target is not responding to interrupt requests.`)

UweBonnes, 2022-01-12:
If your WFI is really at the startup, with no interrupt already set up, WFI has no reason to wake up and probably GDB has no chance to break. Please try again with e.g the timer tick  interrupt set up and activated before executing  WFI. What is your intention of the early WFI? Is it to stop there to wait for the debugger? Connect under reset and set a breakpoint on main would do that task. Or put a breakpoint instruction instead the WFI.

schodet, 2022-01-12:
Setting the CR_SLEEP should allow breaking the MCU even in WFI. I put it here just for the demonstration.

I made an experiment using an older BMP version (v1.6.1-1-g74af1f5, programmed by 1BitSquared I think):

```
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor version
Black Magic Probe (Firmware v1.6.1-1-g74af1f5) (Hardware Version 3)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) mon connect_srst enable
(gdb) mon tpwr enable
(gdb) mon sw
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F4xx
(gdb) att 1
Attaching to program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf, Remote target
reset_handler () at ../../cm3/vector.c:67
67              for (src = &_data_loadaddr, dest = &_data;
(gdb) x/wx 0xE0042004
0xe0042004:     Cannot access memory at address 0xe0042004
(gdb) set mem inaccessible-by-default off
(gdb) x/wx 0xE0042004
0xe0042004:     0x00000000
(gdb) set *(volatile unsigned int *)0xE0042004 = 7
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf 
^C
Program received signal SIGINT, Interrupt.
main () at main.c:47
47              clock_setup();
(gdb) list
42      int main(void)
43      {
44              int i;
45              __asm__ __volatile__ ("wfi");
46
47              clock_setup();
48              gpio_setup();
49
50              /* Set two LEDs for wigwag effect when toggling. */
51              gpio_set(GPIOA, GPIO8);
(gdb)
```

Now with a recent version (v1.7.1-336-g8a09e03, from #973):

```
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon version
Black Magic Probe v1.7.1-336-g8a09e03, Hardware Version 3
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) mon connect_srst enable
Assert SRST during connect: enabled
(gdb) mon tpwr enable
Enabling target power
(gdb) mon sw
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F40x M4
(gdb) att 1
Attaching to program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf, Remote target
0x00000000 in ?? ()
(gdb) x/wx 0xE0042004
0xe0042004:     Cannot access memory at address 0xe0042004
(gdb) set mem inaccessible-by-default off
(gdb) x/wx 0xE0042004
0xe0042004:     Cannot access memory at address 0xe0042004
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf 
^C^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
Disconnected from target.
(gdb)
```

There was a regression somewhere.

DX-MON, 2022-01-13:
Thank you for reporting the Git commit hash of a working version - we will look at bisecting and seeing why and where this changed and if we can reasonably address this regression

UweBonnes, 2022-01-13:
The log shows that the device was running into the WFI() trap while attach under reset was requested. #974 fixes this.

schodet, 2022-01-13:
@UweBonnes thanks for the fix :)

Now with the two PR (cherry-picked):

```
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon assert_srst scan
Target does not support this command.
(gdb) mon connect_srst enable
Assert SRST during connect: enabled
(gdb) mon tpwr enable
Enabling target power
(gdb) mon sw
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F40x M4
(gdb) att 1
Attaching to program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf, Remote target
reset_handler () at ../../cm3/vector.c:67
67              for (src = &_data_loadaddr, dest = &_data;
(gdb) x/wx 0xE0042004
0xe0042004:     Cannot access memory at address 0xe0042004
(gdb) set mem inaccessible-by-default off
(gdb) x/wx 0xE0042004
0xe0042004:     0x00000007
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf 
^C^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
Disconnected from target.
```

Would it be possible that using `run` resets the register?

schodet, 2022-01-13:
Same test, but with:

```c
diff --git a/src/main.c b/src/main.c
index 3a96781..c5b79e3 100644
--- a/src/main.c
+++ b/src/main.c
@@ -42,6 +42,10 @@ static void gpio_setup(void)
 int main(void)
 {
        int i;
+        *(volatile unsigned int *) 0xE0042004 = 7;
+        for (i = 0; i < 10; i++) /* Wait a bit. */
+                __asm__("nop");
+       __asm__ __volatile__ ("wfi");
 
        clock_setup();
        gpio_setup();
```

With this program, I am able to interrupt the MCU.

UweBonnes, 2022-01-13:
Setting DBGMCU_CR probably needs some CPU cycles to propagate through the core and the MCU. With WFI as first command, these cycles are not available and execution hangs uninterruptable (beside NMI) in WFI.  WFI() as first command also does not make sense to me.

What is your intention to use WFI as first command?

Any more need to care for this corner case?

schodet, 2022-01-15:
My intention is just to show the problem, DBGMCU_CR is written by the BMP many cycles before entering main. But OK, I change my program so that it WFI later.

schodet, 2022-01-15:
OK, you are right this makes a difference.

So everything works with #974 and #973.

schodet, 2022-01-15:
The difference may be on the *before* or *after* the `clock_setup`.

UweBonnes, 2022-01-16:
As I understand,your first examples,  WFI() was the very first instruction! The CPU may have set up option bytes, HSI etc before, but I do not think the CPU did the works to set this up, but some state machine did that work. So probably no real CPU cycles where executed and DBGMCU_CR setting had no chance to propagate.

schodet, 2022-01-16:
Before main, there is the code which clears the BSS and copy the DATA, not much, but not zero.

schodet, 2022-02-13:
I think this can be closed, it works with Uwe fixes, still need to merge #973.

closed 2022-02-13 by schodet