blackmagic-debug/blackmagic #250: Can not interrupt 1bitsy when using WFI

schodet, 2017-06-19:
Configuration : BMP 2.1c connected to USB, 1bitsy 1.0d connected using the debug connector, no USB on 1bitsy.

Program is derived from 1bitsy-locm3-template, with the following diff (this WFI after 5 seconds):

```diff
diff --git a/src/main.c b/src/main.c
index e5bc2b5..e94ea09 100644
--- a/src/main.c
+++ b/src/main.c
@@ -50,12 +50,15 @@ int main(void)
        gpio_set(GPIOA, GPIO8);

        /* Blink the LEDs (PA8) on the board. */
+       int j = 0;
        while (1) {
                /* Toggle LEDs. */
                gpio_toggle(GPIOA, GPIO8);
                for (i = 0; i < 6000000; i++) { /* Wait a bit. */
                        __asm__("nop");
                }
+               if (j++ == 5)
+                   while (1) __asm__ __volatile__ ("wfi");
        }

        return 0;
```

Here is gdb log:

```
GNU gdb (7.7.1+dfsg-1+6) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from src/1bitsy-example.elf...done.
(gdb) target extended-remote /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7BAE5BAD-if00
Remote debugging using /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7BAE5BAD-if00
(gdb) mon connect_srst
Assert SRST during connect: disabled
(gdb) mon connect_srst enable
(gdb) mon connect_srst
Assert SRST during connect: enabled
(gdb) mon tpwr enable
(gdb) mon swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F4xx
(gdb) attach 1
Attaching to program: /home/nicolas/work/1bitsy/1bitsy-locm3-template/src/1bitsy-example.elf, Remote target
reset_handler () at ../../cm3/vector.c:68
68              for (src = &_data_loadaddr, dest = &_data;
(gdb) c
Continuing.
^CTimeout sending interrupt, is target in WFI?
^C^CInterrupted while waiting for the program.
Give up (and stop debugging it)? (y or n) y
Quit
(gdb) quit
```

It might be a hardware problem because I can interrupt WFI at work with BMP 2.0. I would be glad if someone can try to reproduce the problem using the same setup.

gsmcmullin, 2017-06-19:
On most Cortex-M targets it is not possible to enter debug mode (core halted) when in WFI or WFE.  This is the default behaviour on STM32, but can be configured.  Check the `DBGMCU_CR` register bits.

I don't see how it is possible that this worked with BMP 2.0 with the same binary loaded on the target.

schodet, 2017-06-20:
You are completely right, my other program set `DBGMCU_CR`.

Sorry for the noise.

closed 2017-06-20 by schodet