blackmagic-debug/blackmagic #259: BMP fails to flash non-padded .rodata section and gets stuck until POR

therealprof, 2017-07-16:
When trying to flash an image which lacks 4-byte padding at the end of an .rodata data the `load` command in `gdb` fails with an error and the BMP is stuck until it is power reset. The target is a STM32F042 connected to the only true BMP (tm) running a recent build.

Here's the debug output of the communication:
```
(gdb) load
Sending packet: $vFlashErase:08000000,00000400#c6...Ack
Packet received: OK
Loading section .vector_table, size 0xc0 lma 0x8000000
Sending packet: $vFlashWrite:8000000:\000\030\000 �\000\000\bE\003\000\bG\003\000\bI\003\000\bK\003\000\bM\003\000\b\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000\000O\003\000\b\000\000\000\000\000\000\000\000Q\003\000\b\001\001\000\b\005\003\000\b\a\003\000\b\t\003\000\b\013\003\000\b\r\003\000\b\017\003\000\b\021\003\000\b\023\003\000\b\025\003\000\b\027\003\000\b\031\003\000\b\e\003\000\b\035\003\000\b\037\003\000\b!\003\000\b}\003\003\000\b%\003\000\b\000\000\000\000\000\000\000\000'\003\000\b\000\000\000\000)\003\000\b+\003\000\b-\003\000\b\000\000\000\000/\003\000\b1\003\000\b3\003\000\b5\003\000\b\000\000\000\0007\003\000\b9\003\000\b#6b...Ack
Packet received: OK
Loading section .text, size 0x334 lma 0x80000c0
Sending packet: $vFlashWrite:80000c0:е\002�\tH\nI\t\032\003"�C!@\000�\212�\aH\bI\n\032"@\aI\000�A�\000�\217�0����F\000\000\000 \004\000\000 \004\000\000 \004\000\000 \004\004\000\b�\003�\203�\001!��\020\200\000\221\b@\001\220r�7K� \002\220\034F`\034!xI�Ͳ\002\232 2R�\221B\001�\004F?�\037&.@\031F\r1\000"\210B\005�`x?"\002@�\034!F\000�\004F�\001в\003C\002\232\225B&�&J\r2\000}\003\221B\004�\nx?}\003\023@L\034"F\200\001ٲ\001C0\003\bC\002\233\0203۲\235B\025�\034K\r3\000 \232B\003�\023x? \030@T\034\a"\222\004�\004\023@\211\001\031CŲ\rC\023K\004�\035F\022K\001�\005F\020K\013 �C\020HAX\200"\021B��\001h\005`\030F\r0\204B��\013H\001x\013J\023F\0203\000)\000�\032F\002}\003\023`\000\232Q@\001p\001\230\000(\000�b�\003���F�\003\000\b(8\001@\000\000\000 \030\004\000H�\003�\202�6H\000x��\020\200r�4J5I\021b\001!\b@\001\220K\004\024j\034B��1L0%%`%h &5B��1%%`\017 \000\220�C\003%\225Q\220Y(@\003(��\020h\030C\020`\210\004\023h\003C\023`&H\006h\f}\003\236C\004}\003\036C\006`�\004\t&�\0064h\204C�\004 C0`(\0054h\204CH\005 C0`pj�}\004�C\020}\004\004Ctb\000\230\000\002tj\204C\b\002\004Ctb\214\003Uh%CU`�0\023J�`\020h\r \020`\000 P`\220`\020H\201`\002h\232C\002`\017JB`\002h\nC\002`\001h\002"\nC\002`\001\230\000(\000�b�\000 \002���F�\000\000\b\024\020\002@\200\000\001\000\000 \002@\000\004\000H\0008\001@\020�\000�\200\215[\000\e�\032�\031�\030�\027�\026�\025�\024�\023�\022�\021�\020�\017�\016�\r�\f�\013�\n�\t�\b�\a�\006�\005�\004�\003�\002�\001�\000�����\b\200����������������\000}\n\005�\013x\003pI\034@\034R\036��pG�\003�\201�\004}\n\017�\003}\003�C\023\037\000\223\035@,\035\013\031\004\031@�@�\022\037\003}\n��\000\230B\e\001�\004F\013F F\031F����\001��\000)\003�\002p@\034I\036��pG�\003�\201�\023\006Ҳ\023C\025\004\035C\024\002,C\004)\013�\003}\003�C\016\0375@C\031\e\035\020�\t\037\003)��q\e\000�\003F\030F����\001��\200�\000�\000"����\200�#a8...Ack
Packet received: OK
Loading section .rodata, size 0xd lma 0x80003f4
Sending packet: $vFlashWrite:80003f4:Hello World!\n#bf...Ack
Packet received: EFF
Error writing data to flash
```

gsmcmullin, 2017-07-16:
The flash write of the .rodata section overflows the erased memory range by 1 byte.  This is likely the cause of failure, and is probably a GDB issue.

Why there is a problem with BMP after this failure still needs to be investigated.

therealprof, 2017-07-16:
File exhibiting the problem:

[serial_helloworld.zip](https://github.com/blacksphere/blackmagic/files/1151412/serial_helloworld.zip)

gsmcmullin, 2017-10-08:
Hopefully the BMP issue is fixed by #293 and #297

closed 2019-03-22 by UweBonnes