blackmagic-debug/blackmagic #262: adiv5_dp_init() doesn't correctly ID APs on Ti Tiva TM4C129X

rbsexton, 2017-07-22:
I was adding support for the Ti Tiva MCU, and I found a problem with detection - 

The probe detects multiple APs during the scan in adiv5_dp_init():

adiv5.c, line 435:

	for(int i = 0; i < 8; i++) {
		ADIv5_AP_t *ap = adiv5_new_ap(dp, i);
		if (ap == NULL)
			continue;

		extern void kinetis_mdm_probe(ADIv5_AP_t *);
		kinetis_mdm_probe(ap);

		if (ap->base == 0xffffffff) {
			/* No debug entries... useless AP */
			adiv5_ap_unref(ap);
			continue;
		}

		/* Should probe further here to make sure it's a valid target.
		 * AP should be unref'd if not valid.
		 */

		/* The rest sould only be added after checking ROM table */
		adiv5_component_probe(ap, ap->base);
	}

This check passes each time - the 'ap->base == 0xffffffff' check isn't working.  The result is that after detecting ~ 34 APs, the probe runs out of memory and crashes.   All of these APs have the same IDR, CFG, BASE, and CSW values:

```
SAM4L: Probe function called
0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace)
0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint)
0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module)
0xe0040000: Debug component - Cortex-M4 TPIU (Trace Port Interface Unit)
0xe0041000: Debug component - Cortex-M4 ETM (Embedded Trace)
AP 14: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040
0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space)
-> cortexm_probe
```

If I work around this problem by limiting the scan to 8 APs I get this:

```
(gdb) monitor jtag_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      TI Stellaris/Tiva
 2      TI Stellaris/Tiva
 3      TI Stellaris/Tiva
 4      TI Stellaris/Tiva
 5      TI Stellaris/Tiva
 .
 .
 .
```

I can attach to the device and it functions normally.

esden, 2021-12-15:
I am closing this issue for now as it seems stale. If this is still a bug in the newest firmware please feel free to reopen this. But a lot has changed in the firmware since this was filed.

closed 2021-12-15 by esden

UweBonnes, 2021-12-15:
There is also a check for " Duplicate base" now since dc3fd2eb0