blackmagic-debug/blackmagic #265: Cannot erase sectors in bank 2 even if they are defined in target flash

Arrowana, 2017-08-21:
I am using the STM32F439 on a custom PCB for debugging.
All functionality work except writing in bank 2. The device is properly detected as a STM32F4xx and has for IDCODE 0x419

The flash can't be programmed in bank 2 even if it is defined in the function **stm32f4_probe(target *t) in stm32f4.c**
I get the following error message:
"Error erasing flash with vFlashErase packet"
https://github.com/blacksphere/blackmagic/blob/master/src/target/stm32f4.c#L189-L191

I am not familiar with the blackmagic probe firmware but can this be an issue related to the fact that the bank 2 is declared before bank 1? Maybe the link.

The current hacky solution I found was to declare bank 2 after bank 1, doing so it is possible to erase sectors in bank2.

UweBonnes, 2017-08-22:
Good catch! When rewriting the F4 code, I did not know that add-flash needs to go bottom up. I think it is not documented neither. Please try https://github.com/UweBonnes/blackmagic if it fixes you problem.

gsmcmullin, 2017-09-19:
The order in which the flash banks are added should not make a difference.  If it does, there is some other bug that needs fixing.

This function selects the flash bank based on address: https://github.com/blacksphere/blackmagic/blob/master/src/target/target.c#L188-L195

I'll try this here and try to debug the underlying issue.

UweBonnes, 2017-09-24:
#285 should fix it. Can you please retest?

Arrowana, 2017-09-27:
I am working with the latest master ad71db05b9c5b167f48f15a42cf9f86255942253
I verified my target IDCODE is 0x419 and https://github.com/blacksphere/blackmagic/blob/master/src/target/stm32f4.c shows proper declaration of flash

Below is a test I did with different intel hex files, the result is very strange

firmware.hex starts at 0x0802 0000, I can load it in flash
```
(gdb) load firmware.hex
Loading section .sec1, size 0x188 lma 0x8020000
Loading section .sec2, size 0xfe00 lma 0x8020200
Loading section .sec3, size 0x10000 lma 0x8030000
Loading section .sec4, size 0x10000 lma 0x8040000
Loading section .sec5, size 0x10000 lma 0x8050000
Loading section .sec6, size 0x10000 lma 0x8060000
Loading section .sec7, size 0x10000 lma 0x8070000
Loading section .sec8, size 0xa3b0 lma 0x8080000
Start address 0x804e67c, load size 435000
Transfer rate: 17 KB/sec, 966 bytes/write.
(gdb) exec firmware.hex
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
(gdb) compare-sections
Section .sec1, range 0x8020000 -- 0x8020188: matched.
Section .sec2, range 0x8020200 -- 0x8030000: matched.
Section .sec3, range 0x8030000 -- 0x8040000: matched.
Section .sec4, range 0x8040000 -- 0x8050000: matched.
Section .sec5, range 0x8050000 -- 0x8060000: matched.
Section .sec6, range 0x8060000 -- 0x8070000: matched.
Section .sec7, range 0x8070000 -- 0x8080000: matched.
Section .sec8, range 0x8080000 -- 0x808a3b0: matched.
```

If I offset this firmare of 0x10 0000, to be at a similar location but in bank 2, it is partially written but wrong at the end.
```
(gdb) load firmware_offset.hex
Loading section .sec1, size 0x188 lma 0x8120000
Loading section .sec2, size 0xfe00 lma 0x8120200
Loading section .sec3, size 0x10000 lma 0x8130000
Loading section .sec4, size 0x10000 lma 0x8140000
Loading section .sec5, size 0x10000 lma 0x8150000
Loading section .sec6, size 0x10000 lma 0x8160000
Loading section .sec7, size 0x10000 lma 0x8170000
Loading section .sec8, size 0xa3b0 lma 0x8180000
Start address 0x814e67c, load size 435000
Transfer rate: 26 KB/sec, 966 bytes/write.
(gdb) exec firmware_offset.hex
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
(gdb) compare-sections
Section .sec1, range 0x8120000 -- 0x8120188: matched.
Section .sec2, range 0x8120200 -- 0x8130000: matched.
Section .sec3, range 0x8130000 -- 0x8140000: matched.
Section .sec4, range 0x8140000 -- 0x8150000: matched.
Section .sec5, range 0x8150000 -- 0x8160000: matched.
Section .sec6, range 0x8160000 -- 0x8170000: matched.
Section .sec7, range 0x8170000 -- 0x8180000: matched.
Section .sec8, range 0x8180000 -- 0x818a3b0: MIS-MATCHED!
warning: One or more sections of the target image does not match
the loaded file
```

If I try to load content directly at 0x0810 0000, I get an error when the probe is trying to erase
```
(gdb) load stuff_at_beginning_of_bank2.hex
Error erasing flash with vFlashErase packet
```

UweBonnes, 2017-09-27:
Pierre,

my patch is still a pull request. Withe the blacksphere repository as upstream, you need to do something like
 git fetch upstream pull/285/head:f4_rework
 git checkout f4_rework

Arrowana, 2017-09-28:
@UweBonnes I tried https://github.com/blacksphere/blackmagic/pull/285
Then wrote a little gdb test script that load intel hex and verify it is has been written properly.
I load files in bank 1 and bank 2 a various locations and it works. Good work, thanks

esden, 2021-12-15:
Thank you @Arrowana based on your comment I understand that the issue is now fixed. And as #285 is merged now I am closing this issue.

closed 2021-12-15 by esden