blackmagic-debug/blackmagic #267: After triggering a "SIGSEGV" on target, target always "running" according to gdb

jmesmon, 2017-08-24:
After connecting to a nrf52 triggering a "SIGSEGV" in target, target is considered always running by gdb and no interaction is allowed. Could be a gdb issue I guess. The actual target appears to not be running at all as the lights aren't flashing. 

GDB is version 8.0, arm-none-eabi-gdb.

```
(gdb) target extended-remote /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7AB796A7-if00
Remote debugging using /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7AB796A7-if00
(gdb) monitor help
General commands:
        version -- Display firmware version info
        help -- Display help for monitor commands
        jtag_scan -- Scan JTAG chain for devices
        swdp_scan -- Scan SW-DP for devices
        targets -- Display list of available targets
        morse -- Display morse error message
        connect_srst -- Configure connect under SRST: (enable|disable)
        hard_srst -- Force a pulse on the hard SRST line - disconnects target
        tpwr -- Supplies power to the target: (enable|disable)
        traceswo -- Start trace capture
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Nordic nRF52
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0001e93a in ?? ()
(gdb) c
Continuing.

Program received signal SIGSEGV, Segmentation fault.
0x00013150 in ?? ()
(gdb) c
Continuing.
Can't send signals to this remote system.  SIGSEGV not sent.
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again.
(gdb) interrupt
(gdb) c
Cannot execute this command while the selected thread is running.
(gdb) inter
interpreter-exec  interrupt         
(gdb) inter
interpreter-exec  interrupt         
(gdb) interrupt 1
(gdb) help interrupt 
Interrupt the execution of the debugged program.
If non-stop mode is enabled, interrupt only the current thread,
otherwise all the threads in the program are stopped.  To 
interrupt all running threads in non-stop mode, use the -a option.
(gdb) monitor help
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again.
```

I have to close that gdb and open a new one to get back in (gdb doesn't want to disconnect because it can't `kill`):
```
(gdb) target extended-remote /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7AB796A7-if00
Remote debugging using /dev/serial/by-id/usb-Black_Sphere_Technologies_Black_Magic_Probe_7AB796A7-if00
(gdb) monitor version
Black Magic Probe (Firmware v1.6.1) (Hardware Version 3)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
```

Now, after reconnecting to the target and using `run`, the blinky lights on the target begin flashing. Hitting `ctrl-c` in gdb results in it reporting that the target is not responding to interrupt requests:
```
(gdb) si
0x0001da9e in ?? ()
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program:
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
The target is not responding to interrupt requests.
Stop debugging it? (y or n) 
Please answer y or n.
The target is not responding to interrupt requests.
Stop debugging it? (y or n) n
The target is not responding to interrupt requests.
Stop debugging it? (y or n) n
```

Saying "y" gets us back to the same advice to use `interrupt` which does nothing:
```
The target is not responding to interrupt requests.
Stop debugging it? (y or n) 
Please answer y or n.
The target is not responding to interrupt requests.
Stop debugging it? (y or n) n
The target is not responding to interrupt requests.
Stop debugging it? (y or n) n
The target is not responding to interrupt requests.
Stop debugging it? (y or n) Quit
(gdb) attach 1
A program is being debugged already.  Kill it? (y or n) y
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again.
(gdb) Quit
(gdb) Quit
(gdb) interrupt
(gdb) 
```

At this point, trying to quit & re-connect with gdb results in the `target` command hanging until killed.

gsmcmullin, 2017-10-09:
The target is generally not in a usable state to resume after SIGSEGV is reported. Resuming with `c` after this will lead to undefined/unpredictable behaviour.

Why it can't be interrupted is most probably unrelated, but there isn't enough information here for me to take a guess as to what's going on.  Please try again with `set debug remote 1` and report the results. That may help by providing a little more insight into the BMP's state.

veerendra-nath, 2018-08-13:
hey i got same error this is the result after setting `set debug remote 1 `
`GNU gdb (GNU Tools for ARM Embedded Processors (Build 17.03)) 7.10.1.20160923-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-atollic-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Sending packet: $qSymbol::#5b...Ack
Packet received: 
Packet qSymbol (symbol-lookup) is NOT supported
Sending packet: $g#67...Ack
Packet received: fa00000000200000008000001b0000009402000000000000d40200200000000000040000d40200200000000004000020884f0020e04f002093070008b402000800000021e04f0020c0d3cccd00000000
Sending packet: $m80002b4,4#5d...Ack
Packet received: 024b1b68
Sending packet: $qL1200000000000000000#50...Ack
Packet received: 
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $vKill;1#6e...Ack
Packet received: 
Packet vKill (kill) is NOT supported
Sending packet: $k#6b...Ack
Sending packet: $?#3f...Ack
Packet received: W00
Sending packet: $vRun;#e6...Ack
Packet received: T05
Packet vRun (run) is supported
Sending packet: $qC#b4...Ack
Packet received: 
Sending packet: $qAttached#8f...Ack
Packet received: 
Packet qAttached (query-attached) is NOT supported
Sending packet: $qOffsets#4b...Ack
Packet received: 
Sending packet: $Hg0#df...Ack
Packet received: 
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/>    <reg name="r11" bitsize="32"/>    <reg name="r12" bitsize="32"/>    <reg name="sp" bitsize="32" type="data_ptr"/>    <reg name="lr" bitsize="32" type="code_ptr"/>    <reg name="pc" bitsize="32" type="code_ptr"/>    <reg name="xpsr" bitsize="32"/>    <reg name="msp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="psp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="primask" bitsize="8" save-restore="no"/>    <reg name="basepri" bitsize="8" save-restore="no"/>    <reg na
Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11...
putpkt: Junk: Ack
Packet received: mme="faultmask" bitsize="8" save-restore="no"/>    <reg name="control" bitsize="8" save-restore="no"/>  </feature></target>
Sending packet: $qXfer:features:read:target.xml:475,3fb#b6...Ack
Packet received: l
Sending packet: $g#67...Ack
Packet received: e80300000020000000800000e80300000000000000000000d40200200000000000040000d40200200000000004000020884f002000500020ffffffff580200080000000100500020c0d3cccd00000000
Sending packet: $qXfer:memory-map:read::0,3fb#e5...Ack
Packet received: m<memory-map><memory type="ram" start="0x20000000" length="0x5000"/><memory type="flash" start="0x08000000" length="0x20000"><property name="blocksize">0x400</property></memory></memory-map>
Sending packet: $qXfer:memory-map:read::bd,3fb#7b...Ack
Packet received: l
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Note: automatically using hardware breakpoints for read-only addresses.
Sending packet: $Z1,8000152,2#75...Ack
Packet received: OK
Packet Z1 (hardware-breakpoint) is supported
Sending packet: $vCont?#49...Ack
Packet received: 
Packet vCont (verbose-resume) is NOT supported
Sending packet: $Hc0#db...Ack
Packet received: 
Sending packet: $c#63...Ack
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again.`

veerendra-nath, 2018-08-13:
`GNU gdb (GNU Tools for ARM Embedded Processors (Build 17.03)) 7.10.1.20160923-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-atollic-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Sending packet: $qSymbol::#5b...Ack
Packet received: 
Packet qSymbol (symbol-lookup) is NOT supported
Sending packet: $g#67...Ack
Packet received: fa00000000200000008000001b0000009402000000000000d40200200000000000040000d40200200000000004000020884f0020e04f002093070008b402000800000021e04f0020c0d3cccd00000000
Sending packet: $m80002b4,4#5d...Ack
Packet received: 024b1b68
Sending packet: $qL1200000000000000000#50...Ack
Packet received: 
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $vKill;1#6e...Ack
Packet received: 
Packet vKill (kill) is NOT supported
Sending packet: $k#6b...Ack
Sending packet: $?#3f...Ack
Packet received: W00
Sending packet: $vRun;#e6...Ack
Packet received: T05
Packet vRun (run) is supported
Sending packet: $qC#b4...Ack
Packet received: 
Sending packet: $qAttached#8f...Ack
Packet received: 
Packet qAttached (query-attached) is NOT supported
Sending packet: $qOffsets#4b...Ack
Packet received: 
Sending packet: $Hg0#df...Ack
Packet received: 
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/>    <reg name="r11" bitsize="32"/>    <reg name="r12" bitsize="32"/>    <reg name="sp" bitsize="32" type="data_ptr"/>    <reg name="lr" bitsize="32" type="code_ptr"/>    <reg name="pc" bitsize="32" type="code_ptr"/>    <reg name="xpsr" bitsize="32"/>    <reg name="msp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="psp" bitsize="32" save-restore="no" type="data_ptr"/>    <reg name="primask" bitsize="8" save-restore="no"/>    <reg name="basepri" bitsize="8" save-restore="no"/>    <reg na
Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11...
putpkt: Junk: Ack
Packet received: mme="faultmask" bitsize="8" save-restore="no"/>    <reg name="control" bitsize="8" save-restore="no"/>  </feature></target>
Sending packet: $qXfer:features:read:target.xml:475,3fb#b6...Ack
Packet received: l
Sending packet: $g#67...Ack
Packet received: e80300000020000000800000e80300000000000000000000d40200200000000000040000d40200200000000004000020884f002000500020ffffffff580200080000000100500020c0d3cccd00000000
Sending packet: $qXfer:memory-map:read::0,3fb#e5...Ack
Packet received: m<memory-map><memory type="ram" start="0x20000000" length="0x5000"/><memory type="flash" start="0x08000000" length="0x20000"><property name="blocksize">0x400</property></memory></memory-map>
Sending packet: $qXfer:memory-map:read::bd,3fb#7b...Ack
Packet received: l
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Sending packet: $m8000152,2#2b...Ack
Packet received: eff3
Note: automatically using hardware breakpoints for read-only addresses.
Sending packet: $Z1,8000152,2#75...Ack
Packet received: OK
Packet Z1 (hardware-breakpoint) is supported
Sending packet: $vCont?#49...Ack
Packet received: 
Packet vCont (verbose-resume) is NOT supported
Sending packet: $Hc0#db...Ack
Packet received: 
Sending packet: $c#63...Ack
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again.`

veerendra-nath, 2018-08-13:
`[nath@GHOST ~]$ arm-none-eabi-gdb 
GNU gdb (GDB) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) tar extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32F1 medium density
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x080002b4 in ?? ()
(gdb) set debug remote 1
(gdb) interrupt
remote_interrupt called
(gdb) c
Continuing.
Sending packet: $vCont?#49...Ack
Packet received: 
Packet vCont (verbose-resume) is NOT supported
Sending packet: $Hc0#db...Ack
Packet received: 
Sending packet: $c#63...Ack
^Cremote_pass_ctrlc called
The target is not responding to interrupt requests.
Stop debugging it? (y or n) `

veerendra-nath, 2018-08-13:
`Attaching to Remote target`
`warning: No executable has been specified and target does not support`
`determining executable automatically.  Try using the "file" command.`
`0x080002b4 in ?? ()`
`(gdb) set debug remote 1`
`(gdb) interrupt`
`remote_interrupt called`
`(gdb) c`
`Continuing.`
`Sending packet: $vCont?#49...Ack`
`Packet received: `
`Packet vCont (verbose-resume) is NOT supported`
`Sending packet: $Hc0#db...Ack`
`Packet received: `
`Sending packet: $c#63...Ack`
`^Cremote_pass_ctrlc called`
`The target is not responding to interrupt requests.`
`Stop`
` debugging it? (y or n) `

xobs, 2020-12-14:
I'm seeing similar issues when trying to bring up an S32K148. I'm also seeing it on the S32K118. It's doing... *something*... to get into a state where gdb simply isn't responding. gdb becomes unresponsive, to the point where the only thing I can do is exit. There is no way to recover the debugging session.

Output from my GDB session:

```
user@Cuboid:~/Code/Ion/gridania-bringup-evt1c/148$ arm-none-eabi-gdb _build/gridania-evt1-148.elf
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from _build/gridania-evt1-148.elf...
(gdb) set debug remote 1
(gdb) tar ext /dev/ttyS7
Remote debugging using /dev/ttyS7
Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Packet instead of Ack, ignoring it
Ack
Packet received: PacketSize=400;qXfer:memory-map:read+;qXfer:features:read+
Packet qSupported (supported-packets) is supported
Sending packet: $vMustReplyEmpty#3a...Ack
Packet received:
Sending packet: $!#21...Ack
Packet received: OK
Sending packet: $Hg0#df...Ack
Packet received:
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: E01
Sending packet: $qTStatus#49...Ack
Packet received:
Packet qTStatus (trace-status) is NOT supported
Sending packet: $?#3f...Ack
Packet received: W00
(gdb) mon swdp_scan
Sending packet: $qRcmd,737764705f7363616e#3c...Ack
Packet received: O54617267657420766f6c746167653a20332e33560a
Target voltage: 3.3V
Packet received: O4b696e6574697320534449443a20307831343866303666613a0a
Kinetis SDID: 0x148f06fa:
Packet received: O4b696e657469732046434647313a20307866663332663030303a0a
Kinetis FCFG1: 0xff32f000:
Packet received: O417661696c61626c6520546172676574733a0a
Available Targets:
Packet received: O4e6f2e20417474204472697665720a
No. Att Driver
Packet received: O20312020202020205333324b313438204d340a
 1      S32K148 M4
Packet received: O20322020202020204b696e65746973205265636f7665727920284d444d2d415029200a
 2      Kinetis Recovery (MDM-AP)
Packet received: OK
(gdb) att 1
Attaching to program: /mnt/d/Code/Ion/gridania-bringup-evt1c/148/_build/gridania-evt1-148.elf, Remote target
Sending packet: $vAttach;1#37...Ack
Packet received: T05
Packet vAttach (attach) is supported
Sending packet: $qC#b4...Ack
Packet received:
Sending packet: $Hg0#df...Ack
Packet received:
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/[508 bytes omitted]
[... 88 lines cut ...]
Sending packet: $c#63...Ack
Packet received: T0B
Sending packet: $g#67...Ack
Packet received: 0400000f000200001ca006404100000010a0064004a0044000a0044004420640000000000000000000000000000000000000000000040020a7060000f806000003000081000400200808002001000004000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Sending packet: $qL1200000000000000000#50...Ack
Packet received:

Program received signal SIGSEGV, Segmentation fault.
Sending packet: $qXfer:memory-map:read::0,3fb#e5...Ack
Packet received: m<memory-map><memory type="ram" start="0x20000000" length="0x1f000"/><memory type="ram" start="0x1ffe0000" length="0x20000"/><memory type="flash" start="0x10000000" length="0x80000"><property name="blocksize">0x1000</property></memory><memory type="flash" start="0x00000000" length="0x180000"><property name="blocksize">0x1000</property></memory></memory-map>
Sending packet: $qXfer:memory-map:read::166,3fb#52...Ack
Packet received: l
Sending packet: $m6f8,4#a1...Ack
Packet received: 4023f9e7
0x000006f8 in main () at src/main.c:111
111             if (c > 96) {
(gdb) stepi
Sending packet: $S0b#e5...Ack
Packet received:
Can't send signals to this remote system.  SIGSEGV not sent.
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again.
(gdb)
```

xobs, 2020-12-14:
I've solved this problem by preventing BMP from sending a SIGSEGV, and instead letting it fall through and sending a SIGTRAP. This prevents gdb from getting into a state where it's unusable.

```patch
diff --git a/src/gdb_main.c b/src/gdb_main.c
index fcf79e6..b77ec26 100644
--- a/src/gdb_main.c
+++ b/src/gdb_main.c
@@ -203,8 +203,8 @@ int gdb_main_loop(struct target_controller *tc, bool in_syscall)
                                gdb_putpacket_f("T%02Xwatch:%08X;", GDB_SIGTRAP, watch);
                                break;
                        case TARGET_HALT_FAULT:
-                               gdb_putpacket_f("T%02X", GDB_SIGSEGV);
-                               break;
+                               // gdb_putpacket_f("T%02X", GDB_SIGSEGV);
+                               // break;
                        default:
                                gdb_putpacket_f("T%02X", GDB_SIGTRAP);
                        }
```

UweBonnes, 2020-12-14:
Any feedback from people with knowledge about gdb is very welcome!

esden, 2021-12-15:
@xobs Does this mean it is a GDB bug? Or is it something that we are not handling right?

xobs, 2021-12-16:
I suspect it's a mismatch on what the `GDB_SIGSEGV` packet actually means. It's been a while since I've run into this problem, since it mostly occurs during early board bringup.

I do suspect it's Blackmagic not handling the fault correctly, but at the moment I don't know enough to be able to debug it further. It seems like a mismatch in state between BMP and GDB, where one thinks the other is halted when it isn't.

esden, 2021-12-16:
Ok, thanks for the info @xobs. As it is hard to reproduce and we don't have a path to address it at the moment I will close this issue for now. If we see it again and are able to reproduce it more reliably maybe we can address it then.

closed 2021-12-16 by esden