blackmagic-debug/blackmagic #280: DFU is not accepting black magic probe's firmware

embhobbb, 2017-09-21:
Hi,
I was trying to flash flash magic probe to stm32f103c8t6. i succeed till some point. I was able to falsh dfu binary. But later dfu is not accepting black magic probe's firmware. 

Here dfu flashed successfully.
`phob@ubuntu:~/blackmagic$ python stm32loader.py -p /dev/ttyUSB0 -e -w -v src/blackmagic_dfu.bin
Bootloader version 22
Chip id: 0x410 (STM32 Medium-density)
Write 256 bytes at 0x8000000
Write 256 bytes at 0x8000100
Write 256 bytes at 0x8000200
Write 256 bytes at 0x8000300
Write 256 bytes at 0x8000400
Write 256 bytes at 0x8000500
Write 256 bytes at 0x8000600
Write 256 bytes at 0x8000700
Write 256 bytes at 0x8000800
Write 256 bytes at 0x8000900
Write 256 bytes at 0x8000A00
Write 256 bytes at 0x8000B00
Write 256 bytes at 0x8000C00
Write 256 bytes at 0x8000D00
Write 256 bytes at 0x8000E00
Write 256 bytes at 0x8000F00
Write 256 bytes at 0x8001000
Write 256 bytes at 0x8001100
Write 256 bytes at 0x8001200
Write 256 bytes at 0x8001300
Write 256 bytes at 0x8001400
Write 256 bytes at 0x8001500
Write 256 bytes at 0x8001600
Write 256 bytes at 0x8001700
Write 256 bytes at 0x8001800
Write 256 bytes at 0x8001900
Write 256 bytes at 0x8001A00
Read 256 bytes at 0x8000000
Read 256 bytes at 0x8000100
Read 256 bytes at 0x8000200
Read 256 bytes at 0x8000300
Read 256 bytes at 0x8000400
Read 256 bytes at 0x8000500
Read 256 bytes at 0x8000600
Read 256 bytes at 0x8000700
Read 256 bytes at 0x8000800
Read 256 bytes at 0x8000900
Read 256 bytes at 0x8000A00
Read 256 bytes at 0x8000B00
Read 256 bytes at 0x8000C00
Read 256 bytes at 0x8000D00
Read 256 bytes at 0x8000E00
Read 256 bytes at 0x8000F00
Read 256 bytes at 0x8001000
Read 256 bytes at 0x8001100
Read 256 bytes at 0x8001200
Read 256 bytes at 0x8001300
Read 256 bytes at 0x8001400
Read 256 bytes at 0x8001500
Read 256 bytes at 0x8001600
Read 256 bytes at 0x8001700
Read 256 bytes at 0x8001800
Read 256 bytes at 0x8001900
Read 256 bytes at 0x8001A00
Verification OK`

Windows recognized as follows.
[https://drive.google.com/file/d/0B6CAhHiAfKJkZzd1NEdfMThmSUE/view?usp=sharing](url)

later i tried to flash probe's binary i got error as [b]invalid dfuse address: 0x08002000:leave[/b]

here some steps i followed
`phob@ubuntu:~/blackmagic$ dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D blackmagic.bin
invalid dfuse address: 0x08002000:leave

phob@ubuntu:~/blackmagic/src$ sudo dfu-util -d 1d50:6018,:6017 -s 0x08002000 -D blackmagic.bin
dfu-util 0.5

(C) 2005-2008 by Weston Schmidt, Harald Welte and OpenMoko Inc.
(C) 2010-2011 Tormod Volden (DfuSe support)
This program is Free Software and has ABSOLUTELY NO WARRANTY

dfu-util does currently only support DFU version 1.0

Filter on vendor = 0x1d50 product = 0x6018
No DFU capable USB device found
`

UweBonnes, 2017-09-21:
You try to flash above the 64 k limit. On a STM32F103C8 chip there many not be flash or the flash is tested bad or not tested at all. Stock dfu-util cares for flash size given in the DFU descriptor. Try <blacksphere>/scripts/stm32_mem.py that does not care for the flash size given in the DFU descriptors on your own risk. Good luck!

devanlai, 2017-09-21:
I think even before running into the 64k limit, the problem dfu-util 0.5 is just ancient. It doesn't seem to recognize the `:leave` command, and it also doesn't seem to really work with the ROM dfuse bootloader I have either.

closed 2017-09-21 by embhobbb