blackmagic-debug/blackmagic #290: How to COMPLETELY erase nrf51?

dmikhalsky, 2017-09-28:
am trying to reprogram a commercial nrf51 module. Under "commerical" I mean "not from aliexpress", but installed in "smart" bluetooth socket.
I can connect to it with Black Magic Probe just fine, but after mon erase_mass, BMP reports 0xfffffffe instead of expected 0xffffffff, meaning something is left behind. That prohibits reflashing of softdevice (load softdevice_filename.hex results in error) BUT I can load "plain" sketches via OpenOCD that do not require softdevice.

I ASSUME the module has some kind of write protection and/or UICR registers set that are not erased with BMP mass erase command. 

From Series Reference Manual from chapter UICR "The code area can be divided into two regions, code region 0 (CR0) and code region 1 (CR1). Code region 0 starts at address 0x00000000 and stretches into the code area as specified in the CLENR0 register. The area above CLENR0 will then be defined as code region 1. If CLENR0 is not configured, that is, has the value 0xFFFFFFFF, the whole code area will be defined as code region 1 (CR1).
So if I understand correctly, BMP tries to load softdevice into CR0 region and fails.

So what's the proper way to really completely erase the module and/or load softdevice?

dmikhalsky, 2017-09-28:
Ok, sorry! But why then I am unable to flash the softdevice? It works OK with another nrf51 chip I have (different module). 
In both cases I make "mon erase_mass" followed by "load softdevice.hex", but it works in one module and doesn't work with another

gsmcmullin, 2017-10-02:
Use `compare-sections` to verify the loaded image.  I have no reason to believe that the erase isn't working, so I'm closing the issue.  If verification fails, please reopen.

closed 2017-10-02 by gsmcmullin